galliot-us/neuralet #2: Only runnable on Coral Dev Board? (As of now)

ROS-lover, 2020-03-17:
Hey,

Thanks for the well-documented and useful repository. I was wondering if the docker images and instructions presented are runnable on Coral Dev Boards only? I followed the instructions but cannot build or run the docker image and demo. 

Ubuntu 18 - Cuda 10.2.

Thanks.

* 1 *
docker run -it --privileged --net=host -v $(pwd)/../../:/repo neuralet/coral-dev-board:$MODEL_NAME
Unable to find image 'neuralet/coral-dev-board:mobilenet_ssd_v2_face' locally
mobilenet_ssd_v2_face: Pulling from neuralet/coral-dev-board
77b067f4ec14: Already exists 
f55f20663706: Already exists 
6cf10d87432d: Already exists 
5209673823c9: Already exists 
853b3604bd0a: Already exists 
48ecfc0f64d8: Already exists 
641c7ec0d6e7: Pull complete 
Digest: sha256:550c93a4aad00967c163df01ecaa0694f4621b4e06a4bcb1cd2cd066571bb5d6
Status: Downloaded newer image for neuralet/coral-dev-board:mobilenet_ssd_v2_face
standard_init_linux.go:211: exec user process caused "exec format error"

*2*
docker build -t "neuralet/coral-dev-board:$MODEL_NAME" .
Sending build context to Docker daemon  30.72kB
Step 1/9 : FROM arm64v8/debian:buster
 ---> 798e532a0a9f
Step 2/9 : VOLUME  /repo
 ---> Using cache
 ---> ba56834da20d
Step 3/9 : RUN apt-get update && apt-get install -y wget gnupg     && rm /etc/apt/sources.list  && rm -rf /var/lib/apt/lists     && wget -qO - https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add -
 ---> Running in 6addf734ef9a
standard_init_linux.go:211: exec user process caused "exec format error"
The command '/bin/sh -c apt-get update && apt-get install -y wget gnupg     && rm /etc/apt/sources.list  && rm -rf /var/lib/apt/lists     && wget -qO - https://packages.cloud.google.com/apt/doc/apt-key.gpg | apt-key add -' returned a non-zero code: 1

mhejrati, 2020-03-17:
Hey there, 

The docker containers are currently built for ARM processors to run on the coral dev board. We plan to release them for x86 so that they can use the other form factors of edge TPU soon, but no exact timeline available yet. We are also planning to create ROS nodes for these models so it is easy to integrate with ROS systems either on ARM or x86. However, keep in mind that all of them require an edge TPU to run. Is that your intended use case?

ROS-lover, 2020-03-19:
Thanks Mohsen. It's more clear now. Looking forward to you ROS devs.

mhejrati, 2020-03-29:
I just released amd64 examples with support for ROS. Please check and let me know if you have any further questions.

rk68, 2020-03-31:
@ROS-lover could you review the last comment and let us know if you have any further questions.

Thanks,
Reza

closed 2020-04-08 by mhejrati

ROS-lover, 2020-08-08:
Hi Reza, I have lost my gitub credentials and recently found your responses. Thank for the them and I will be reviewing the commit soon. Cheers

commit 9127b9f8 (2020-04-08): debugged PR, was breaking the code, added centroids, updated dockerfiles, seems working but have to visualize to understand if it works right. closes #2
commit 7c06e18d (2021-01-15): debugged PR, was breaking the code, added centroids, updated dockerfiles, seems working but have to visualize to understand if it works right. closes #2