galliot-us/neuralet #4: Build a physical distancing detection model to run on Coral Edge TPU

mhejrati, 2020-04-05:
An application that detects pedestrians and identifies if two or more people are in close proximity and violate the physical distancing orders. 

[image]

JsonSadler, 2020-04-05:
Hi @mhejrati 

Great work! We tested your docker files and seems they all are working properly. There are lots of ideas that can be added, but I was going to do the exact same thing as you issued (seems I have accidentally happened to see this issue just when you created it :D about 8 hours ago). 

Me and one of my friends are actively working on this, I could not make a fork as your repo was big, we are using your mobilenet_ssd_v2 here:
[https://github.com/JsonSadler/neuralet-distancing](https://github.com/JsonSadler/neuralet-distancing)
We would be glad if you could take a look at our code base and let us know if you have any comments, and hope you do not mind that we named it after your repo. I actually was thinking of suggesting it as a template application one can develop using your samples (they seem to be developed for super-easy demonstration, lacking a practical demo).

Feel free to contribute, test, and give us feedback.

JsonSadler, 2020-04-05:
Okay just deleted that previous repo and made a fork of yours, working on [this](https://github.com/JsonSadler/neuralet/tree/social-distancing) branch.

mhejrati, 2020-04-06:
Thanks @JsonSadler, this is awesome :)

I will review soon.

mhejrati, 2020-04-09:
Thanks for many contributions @JsonSadler @rk68 @alpha-carinae29 @mrn-mln 

Just created a roadmap here:
https://github.com/neuralet/neuralet/projects/1

HHIR, 2020-04-11:
That's really interesting, how can I run it in a normal system without the extra device ( Ubuntu )?

mhejrati, 2020-04-11:
Hey @HHIR, thanks for reaching out. 

We haven't released that yet, I created an issue #31 for it. You can track progress there.

mhejrati, 2020-05-01:
@HHIR you can now run the app on any x86 system without an accelerator.  Thanks to @kkrampa for submitting #43

mhejrati, 2020-05-07:
I have posted a short presentation about the project and our roadmap here:
https://docs.google.com/presentation/d/13EEt4JfdkYSqpPLpotx9taBHpNW6WtfXo2SfwFU_aQ0/edit?usp=sharing

closed 2022-12-12 by mhejrati