galliot-us/neuralet #30: Improve detection box filtering strategies

mhejrati, 2020-04-09:


alpha-carinae29, 2020-05-16:
@mhejrati please move this issue to fine-tuning and calibration milestone

alpha-carinae29, 2020-05-16:
I worked on this issue in last days.
I have implemented a background subtraction with `cv.createBackgroundSubtractorMOG2` module.
here is a result on Oxford Town Center:
[image]
[image]
Now I am trying to filter object detection's output based on this information. To doing this I will filter boxes that the percentage of background pixels more than a threshold.

alpha-carinae29, 2020-05-16:
Also I am implementing a prior distribution of bounding boxes for each pixel in the image. (only mean and variance not the whole distribution.) 
To this end I computed the average and variance of width and height of bounding boxes for the centroids of the boxes recurrently with [Welford algorithm](https://en.wikipedia.org/wiki/Algorithms_for_calculating_variance#Welford's_online_algorithm). After the training Phase, for each detected bounding box we compare its width and height with the prior mean and variance and if we detect an outlier we eliminate the bounding box.

alpha-carinae29, 2020-05-16:
@mhejrati please mark it as "in progress".

closed 2022-12-12 by mhejrati