galliot-us/neuralet #38: cv2 error

mrn-mln, 2020-04-19:
After ~2 minutes of running below error is raised.

Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 16, in __init__
    self.ui.start()
  File "/repo/applications/smart-distancing/ui/web_gui.py", line 149, in start
    self.__ENGINE_INSTANCE.process_video(video_path)
  File "/repo/applications/smart-distancing/libs/core.py", line 78, in process_video
    cv_image, objects, distancings = self.__process(cv_image)
  File "/repo/applications/smart-distancing/libs/core.py", line 49, in __process
    cv_image = cv.resize(cv_image, tuple(resolution))
cv2.error: /build/opencv-ytxPbe/opencv-3.2.0+dfsg/modules/imgproc/src/imgwarp.cpp:3492: error: (-215) ssize.width > 0 && ssize.height > 0 in function resize

mhejrati, 2020-04-20:
@mrn-mln what device did you run this on?

mrn-mln, 2020-04-20:
@mhejrati I ran on AMD64 node with a connected Coral USB Accelerator.

mrn-mln, 2020-04-20:
@mhejrati I think this bug is raised when there are no frames (the frame streams is ended) as a result cv_image shape is ()
cv2.resize((), tuple(resolution)) # ERROR
I think you should check the frame's shape before calling the `__process` method.
I fixed the issue by considering the shape of the frame before processing. 

        while input_cap.isOpened() and self.running_video:
            _, cv_image = input_cap.read()
            if np.shape(cv_image) != ():
                cv_image, objects, distancings = self.__process(cv_image)
            else:
                continue

mrn-mln, 2020-04-25:
I closed this issue since [it solved](https://github.com/neuralet/neuralet/pull/41/commits/fa9e6ddf89168990c8b5b3e337d069e95cdf1e14).

closed 2020-04-25 by mrn-mln