galliot-us/neuralet #44: Error running social-distance

ticlazau, 2020-04-25:
Hello, 

I am trying to run the social-distance like this:

docker run -it --runtime nvidia --privileged -p 8000:8000 -v /data/projects/neuralet:/repo neuralet/jetson-nano:applications-smart-distancing

But I am getting this error:
'
Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 13, in __init__
    self.engine = CvEngine(self.config)
  File "/repo/applications/smart-distancing/libs/core.py", line 19, in __init__
    self.logger = Logger(self.config)
  File "/repo/applications/smart-distancing/libs/loggers/loggers.py", line 19, in __init__
    self.name = self.config.get_section_dict("Logger")["Name"]
  File "/repo/applications/smart-distancing/libs/config_engine.py", line 59, in get_section_dict
    return self.section_options_dict[section]
KeyError: 'Logger'

'

Any ideas how I can overcome this issue?

Thank you,
FM

SaddamBInSyed, 2020-04-25:
Hi.

App try to find 'Logger' field in the config file but lo luck.

You can find the same on another ' config- skeleton.ini file ' and paste into the condig-jetson file.

Hope this helps.

alpha-carinae29, 2020-04-25:
Hi @ticlazau , thanks for your feedback, as @SaddamBInSyed mentioned you should copy:
```
[Logger]
Name: csv_logger
TimeInterval: 0.5
LogDirectory: /repo/applications/smart-distancing/ui/static/data
```
to the bottom of the `config-jetson.ini` file

ticlazau, 2020-04-25:
Thank you! 
all good now, I am able to start and run the video example.

Rgds
FM

PS: with other videos social distance is not working, event with two persons in the video

closed 2020-04-25 by ticlazau

alpha-carinae29, 2020-04-25:
@ticlazau would you please open another issue and describe the problem with other video with more details?
Tnx

PR #46 (2020-04-26): fixed #44
added logger configs to Jetson config file.
this will close #44
commit 5ad91aab (2020-04-26): Merge pull request #46 from alpha-carinae29/add-logger-config-jetson

fixed #44
commit 5c6c9b6c (2020-04-26): fix #44