galliot-us/neuralet #47: Face mask detection detector

mhejrati, 2020-04-26:
Curate a dataset and build a facemask detection model.

A sample dataset can be found here:
https://github.com/VirajDeshwal/mask-detection-dataset

SaddamBInSyed, 2020-04-30:
@mhejrati 
Hi.

I have a face mask detection model (trained on tfod 1.x) which will give good fps on cpu.

if you want i can share the same. if satisfied then you can integrate this option into the main repo.

thanks.

linux-devil, 2020-05-01:
There is one face mask detection model available on padddlehub

mhejrati, 2020-05-01:
@SaddamBInSyed yes, it will be great to test it. What dataset did you train and test it on?

mhejrati, 2020-05-01:
@linux-devil can you please provide the link?

SaddamBInSyed, 2020-05-01:
@mhejrati 

The dataset prepared locally with 10k images by my known person.
This was tested with both webcam and CCTV footage and seems works fine.

Mainly this was developed bcz this model works fine with CPU.

Also I came across this below repo who has model for face mask detection

https://github.com/AIZOOTech/FaceMaskDetection

Please have a look and let me know your comments.

Thanks

linux-devil, 2020-05-01:
@mhejrati There you go : https://www.paddlepaddle.org.cn/hub/scene/maskdetect 
Its pretty good and lightweight , I already tested this model . Not sure how to integrate it here. 
Above repo was open sourced by Baidu .

mhejrati, 2020-05-08:
@SaddamBInSyed @linux-devil Thanks a lot for the pointers. Please keep your suggestions about new datasets here. 

I am hoping we can crowdsource an open and free large facemask dataset for the community. Looking for help and partners there, please let me know if you have any thoughts.

mhejrati, 2020-05-10:
Thanks to @mrn-mln for submitting a PR for this :) I will review. 

BTW, we should start looking to crowdsource some larger scale dataset for it. Does anyone have thoughts?

alpha-carinae29, 2020-05-11:
You Can create a page in [neuralet.com ](neuralet.com)and invite people to send their own images with mask.
Also the user should sign an agreement that confirms that the images are their own images and by signing this he/she give the community the right to use these images for mask detection.

mrn-mln, 2020-05-15:
@mhejrati , I implemented a Mobilenet-SSD version for face detection and integrate it with a [face-mask classifier ](`https://github.com/neuralet/neuralet/pull/68`). The detector detects faces and the detected face is cropped from the original image, and the classifier classifies it to face and mask-face class.
I also used a fine-tune Mobilenet-SSD face detector [[1]](https://github.com/yeephycho/tensorflow-face-detection). For both models [WIDERFACE dataset](http://mmlab.ie.cuhk.edu.hk/projects/WIDERFace/) is used for training.
Moreover, another Mobilenet-SSD has trained with a composed dataset of [WIDER Face](http://mmlab.ie.cuhk.edu.hk/projects/WIDERFace/) and [MAFA](www.escience.cn/people/geshiming/mafa.html). I evaluated all models and this is the result

`
             **Model                               |  Detector Classes  |      |BBox Confidence Threshold|    | IOU  |   |   mAP  |**    

Mobilenet_SSD_V1_Face                      |        Face        |      |           0.50          |    | 0.50 |   | 67.93% |    
SSD_MobileNet_V1_Face [1]                        |        Face        |      |           0.50          |    | 0.50 |   | 64.71% |
Mobilenet_SSD_V1_Face_Mask                 |     Face, Mask     |      |           0.50          |    | 0.50 |   | 76.74% |
Mobilenet_SSD_V1 + Face_Mask Classifier[2] |        Face        |      |           0.50          |    | 0.50 |   | 68.87% |
SSD_MobileNet_V1 [1] + Face_Mask Classifier[2]   |        Face        |      |           0.50          |    | 0.50 |   | 67.26% |

**Evaluation Set:**
 Total Number of Images = 2818
 Total Number of Faces = 2020 
 Total Number of FaceMask = 798

[1] https://github.com/yeephycho/tensorflow-face-detection
[2] https://github.com/neuralet/neuralet-models/tree/master/amd64/face_mask_classifier 
`
Based on the results Mobilenet_SSD_V1_Face_Mask has the best results within all implemented models.

mrn-mln, 2020-05-24:
@mhejrati I sent [a PR](https://github.com/neuralet/neuralet-models/pull/5) including detector models frozen graph for x86 devices and a quantized tflite model for edgetpu devices at the neuralet-model repo.

keivan68, 2020-06-22:
I have prepared a face mask dataset with three classes:
no_mask, improper_mask, proper_mask

The improper_mask class represents the examples where the mask does not fully cover the nose and/or mouth. The dataset also includes bandanas and untypical masks for training more accurate models. 

I will contribute the dataset to the repo after finalizing a few things.

closed 2022-12-12 by mhejrati