galliot-us/neuralet #50: Deepstrem Model

rk68, 2020-04-28:


rk68, 2020-04-28:
Escalated 4/28/2020

mdegans, 2020-04-28:
Y'all need help? Happy to volunteer.

rk68, 2020-04-28:
@mdegans That's great! What do you need to start? Really appreciate the offer.

mdegans, 2020-04-29:
So, I'm looking at your `Detector` interface here and the inference methods expect a resized rgb image. DeepStream works best when it can handle the whole pipeline and everything stays on the GPU, so it would probably be easiest if the `cv.VideoCapture` bit was moved inside the `Detector` class for coral. It also works best when it handles multiple sources at once so the frames can be batched (temporal batching is unsupported at the moment), so I'm thinking `Detector` could be initialized with a list of uris. The Detector base class could look something like this:

```python
class Detector:
    """
    Detector class...
    :param config: a path to the .ini config file for the Detector
    :param uris: a Sequence of uris to use as sources
    :param on_detection_cb: a callback to be called on each frame
    """
    def __init__(self, config, uris):
        self.config = config
        self.uris = uris
        ...

    @property
    def uris(self):
        """returns a list of active uris"""

    @sources.setter
    def uris(self, uris):
        """sets the active uris, adding source elements as necessary""

    def on_buffer(self, detections):
        """
        Callback for every frame.
        This should calculate distances and update the OSD and/or web ui.
        :param detections: a Sequence of some detection class with info about the tracked objects
        """
```
Lastly, the Web UI would probably be most effecient for DeepStream if using a html5 video tag. If you want, I can just fork the project and start work on this. I don't really need anything else to start other than some approval that you're ok with this basic design. If I fork it and do this work, I'd rather it be reintegrated at some point. FYI, I do have coral modules (usb, and pci-e), so I can test it works with both.

I appreciate your minimalist, clean approach to code. Let me know what you think of these changes or if you have other ideas/concerns/whatever before I start.

rk68, 2020-04-29:
@ebertrams, Yes we have TesnorRT AI model and we are working to release more models.

mhejrati, 2020-04-29:
@mdegans 

Thanks a lot for the thoughtful suggestion. I like your suggested approach, I appreciate it if you can work on it. I don't think an integration down the line will be very challenging since we are not working on things that would have a large impact on this.

One question for you to think about while building the system, what is a good design if we want to run parallel pipelines? E.g. in #47 we want to add facemask detection, which requires running 1 or 2 more models, a face detector, and mask classifier.  

Thanks,

mdegans, 2020-04-29:
@mhejrati 

Running multiple models at the same time isn't a problem. DeepStream's nvinfer element supports this in secondary classifier mode. The pipeline is still serial, but buffers are passed down the pipeline without blocking, so work is done on separate CUDA streams and/or devices in parallel. This is not supported for detection models, but multiple classifiers can be run after a the initial detection in this async mode. In any case, the whole pipeline will (hopefully) be encapsulated in the `Detector` object. with a common interface for all devices.

Fwiw Google Coral also has some working, very good, [gstreamer code](https://coral.googlesource.com/edgetpuvision/+/refs/heads/master), so it may be advantagous to just use GStreamer inside each `Detector` object. The License is also Apache. It seems there are positive reactions to the idea, so I will fork this project and start work. I will post progress reports here.

mhejrati, 2020-04-29:
@mdegans great. 

Are you planning to try gstreamer for both jetson and Coral or try DeepStream for Jetson for now?  Please let me know so that I plan the other development accordingly. 

Many thanks,

mdegans, 2020-04-29:
@mhejrati 

Yw. My grandmother's funeral was today (she died of Covid last week), so this is personal to me. I'm happy to help.

**Re: DeepStream**: It's just Nvidia's name for a set of plugins for GStreamer that perform inferences on video pipelines. Google provides code to do more or less the same thing, also using GStreamer as a backend.

It may make sense to use a common backend like GStreamer rather than maintain multiple backends, but my goal _right now_ is _simply to provide common interfaces (eg. `Detector`, `Network` no matter what the backend is_. I will focus on Jetson first, adapting your current code for Coral to the new interface.

Currently I am reviewing your codebase and moving some things around to accommodate these changes. I will push my code and update here later today or tomorrow so you and others on the team can review.

mhejrati, 2020-04-29:
@mdegans very sorry to hear that! I hope everyone else in your family is healthy and safe.

Re GStreamer, it makes sense, looking forward to it.

rk68, 2020-04-29:
> Yw. My grandmother's funeral was today (she died of Covid last week), so this is personal to me. I'm happy to help.
>
> **Re: DeepStream**: It's just Nvidia's name for a set of plugins for GStreamer [...]

@mdegans very sorry to hear that! I appreciate your contribution here.

valdivj, 2020-04-30:
@mdegans 

Those we love don't go away;
They walk beside us every day.

Unseen, unheard but always near.
Sill loved, still missed, and very dear.

Wishing us hope in the midst of sorrow,
Offering comfort in the midst of pain, both today and tomorrow

I'm not the most skilled programmer but I'm pretty good at getting hardware to work together.
I am going to play around to see whats the best camera setups.

mdegans, 2020-05-01:
Thanks for the kind wishes. I haven't had as much time as I wanted over the past few days, but I did get a chance to read more or less everything add some package / test / docs boilerplate, and move some code around. Notable changes:

* [sphinx docs](https://github.com/mdegans/neuralet/tree/gstreamer/applications/smart-distancing/docs) boilerplate which builds the docs, formatting issues aside.
* create a [base class for Detector](https://github.com/mdegans/neuralet/blob/gstreamer/applications/smart-distancing/smart_distancing/detectors/__init__.py#L23), as well as for the [Distancing app itself](https://github.com/mdegans/neuralet/blob/gstreamer/applications/smart-distancing/smart_distancing/core/_distancing.py#L11). Now the [implementation](https://github.com/mdegans/neuralet/blob/gstreamer/applications/smart-distancing/smart_distancing/detectors/edgetpu/_detectors.py#L103) of a Detector looks something like this:
```
class MobilenetSsdDetector(EdgeTpuDetector):
    __doc__ = TFLITE_CLASS_DOC
    DEFAULT_MODEL_FILE = 'mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite'
```
in addition to moving code to a parent, the ABC also enforces consistency across the different implementations.
* I saw you wrote your own logging class, but since I'm very familiar with the python logging module, i wrote some alternative handlers ([json](https://github.com/mdegans/neuralet/blob/gstreamer/applications/smart-distancing/smart_distancing/loggers/_jl_handler.py) and [csv](https://github.com/mdegans/neuralet/blob/gstreamer/applications/smart-distancing/smart_distancing/loggers/_csv_handler.py)) and doctests for them. If you hate the logging module, that's fine too, it's your project, but the logging module is thread/process safe and I'm toying with the idea of running gstreamer in a separate process, so it might come in handy.
* add a `setup.py`, a bunch of `__init__.py`, and other python package boilerplate. It's not quite ready for pypi, but it's getting there.
* add a __main__.py and moved the entrypoint in there. On pip/setup.py install, a script is generated to launch it. The module itself can also be run this way with `python -m smart_distancing`
* support for user data directory in `~./smart_distancing`
* other changes in the git history.

It's more than you expect and perhaps too much, so let me know if anything goes against your design/philosphy/whatever. Fwiw I think I'm done making major changes. Next, I plan on polishing the documentation, writing tests, and then writing the implementation of the Detector, etc... for Jetson.

Thoughts?

mhejrati, 2020-05-01:
@mdegans I just checked your fork and it looks like a huge upgrade :) Thanks a lot!

- I think the new base class for the detector looks good.
- It is fine using the python logging module, all we care about is the flexibility to log various perception outputs with timestamps so that the downstream processes and UI can consume them. I don't see an issue moving it to the python logging module.
- it is cool to prep for pypi release, we were not planning for that but why not ;) 
- Adding main.py makes sense.
- other changes look fine as well.

This is definitely more than I expected, but definitely a welcome upgrade. To make merging more streamlined, do you think you can make a PR before adding DeepStream? I was hoping to merge #43 today, but #51 can be merged after your PR is merged.

mdegans, 2020-05-01:
Thanks for the feedback. I will try to submit a PR by end of day. I want to get things in a working state first. I made a bunch of changes so I want to test things work. First, I need to a new usb-c cable for my Coral since the one I am using is flaky. Worse comes to worse I can yank out the wifi card from my workstation and stick in the PCI version to test on x86.

rk68, 2020-05-01:
@mdegans  Awesome! Looking forward to it.

mdegans, 2020-05-02:
I'm slowly getting there, but not quite ready for a pr. I need to test the coral x86 still.
- setup.py works and the packages installs as expected with all the data. 
- added some code to find the config file and an order or priority for the config paths to make it easier to test. available config files are listed with --help and an absolute path is not required
- added a bunch of debug logging code. The Logging section is handled again and you can specify 'csv' or 'json'. I tested that works, but the file rotation needs testing. It should work.
- merged #43 into fork and gstreamer branch to make it easier for you. based on the log x86 it seems to be working but I haven't pointed my browser at it yet.

```
 $ pip uninstall -y smart-distancing && pip3 install --no-cache . && smart-distancing --verbose --config x86.ini
Found existing installation: smart-distancing 0.1.0
Uninstalling smart-distancing-0.1.0:
  Successfully uninstalled smart-distancing-0.1.0
Processing /home/username/Projects/neuralet/applications/smart-distancing
Requirement already satisfied: Flask in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from smart-distancing==0.1.0) (1.1.2)
Requirement already satisfied: numpy in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from smart-distancing==0.1.0) (1.18.3)
Requirement already satisfied: scipy in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from smart-distancing==0.1.0) (1.4.1)
Requirement already satisfied: Pillow in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from smart-distancing==0.1.0) (7.1.2)
Requirement already satisfied: itsdangerous>=0.24 in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from Flask->smart-distancing==0.1.0) (1.1.0)
Requirement already satisfied: click>=5.1 in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from Flask->smart-distancing==0.1.0) (7.1.2)
Requirement already satisfied: Jinja2>=2.10.1 in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from Flask->smart-distancing==0.1.0) (2.11.2)
Requirement already satisfied: Werkzeug>=0.15 in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from Flask->smart-distancing==0.1.0) (1.0.1)
Requirement already satisfied: MarkupSafe>=0.23 in /home/username/Projects/neuralet/venv/lib/python3.7/site-packages (from Jinja2>=2.10.1->Flask->smart-distancing==0.1.0) (1.1.1)
Building wheels for collected packages: smart-distancing
  Building wheel for smart-distancing (setup.py) ... done
  Created wheel for smart-distancing: filename=smart_distancing-0.1.0-py3-none-any.whl size=37640 sha256=16f7f6e9515991ea863f4e47de5379bf41e5ae5c9a599316cd0c4cafb1459ec0
  Stored in directory: /tmp/pip-ephem-wheel-cache-_xpj45j7/wheels/9d/20/ed/4de4dd311a8e392a48e3c605a370153d1e149e2a1dca41736e
Successfully built smart-distancing
Installing collected packages: smart-distancing
Successfully installed smart-distancing-0.1.0
DEBUG:smart_distancing.__main__:main("x86.ini")
DEBUG:smart_distancing.__main__:load_config:config selected: x86.ini
DEBUG:smart_distancing.__main__:trying /home/username/.smart_distancing/configs
DEBUG:smart_distancing.core._config_engine:ConfigEngine:loading config from /home/username/.smart_distancing/configs/x86.ini
DEBUG:smart_distancing.__main__:trying: /home/username/Projects/neuralet/venv/lib/python3.7/site-packages/smart_distancing/data/config
DEBUG:smart_distancing.core._config_engine:ConfigEngine:loading config from /home/username/Projects/neuralet/venv/lib/python3.7/site-packages/smart_distancing/data/config/x86.ini
DEBUG:smart_distancing.core._distancing:__init__ start
INFO:smart_distancing.core._distancing:device is: x86
INFO:smart_distancing.core._distancing:image size: (300, 300, 3)
2020-05-01 17:42:48.932821: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer.so.6'; dlerror: libnvinfer.so.6: cannot open shared object file: No such file or directory
2020-05-01 17:42:48.932945: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer_plugin.so.6'; dlerror: libnvinfer_plugin.so.6: cannot open shared object file: No such file or directory
2020-05-01 17:42:48.932955: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:30] Cannot dlopen some TensorRT libraries. If you would like to use Nvidia GPU with TensorRT, please make sure the missing libraries mentioned above are installed properly.
INFO:tensorflow:Saver not created because there are no variables in the graph to restore
2020-05-01 17:42:50.767308: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcuda.so.1'; dlerror: libcuda.so.1: cannot open shared object file: No such file or directory
2020-05-01 17:42:50.767330: E tensorflow/stream_executor/cuda/cuda_driver.cc:351] failed call to cuInit: UNKNOWN ERROR (303)
2020-05-01 17:42:50.767362: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:156] kernel driver does not appear to be running on this host (skull): /proc/driver/nvidia/version does not exist
2020-05-01 17:42:50.767591: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2020-05-01 17:42:50.790581: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 3099995000 Hz
[... cut ...]
OpenCV(4.2.0) /io/opencv/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): /repo/applications/smart-distancing/data/TownCentreXVID.avi in function 'icvExtractPattern'

failed to load video  /repo/applications/smart-distancing/data/TownCentreXVID.avi
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1307, in _shutdown
    lock.acquire()
KeyboardInterrupt
```

I'm not sure if I'll have a PR by end of day as I'd hoped. We'll see after I walk my dogs and I do some more testing.

mdegans, 2020-05-02:
Got it working on x86 in software tensorflow. #43 works. Still haven't tested the Coral yet. Gonna commit now and finish the rest tomorrow. Will probably also change around CvDistancing a bit if it doesn't bother you since some methods are called by a callback now.

[image]

mhejrati, 2020-05-02:
Nice :)

mdegans, 2020-05-02:
I fixed the log spam issue and got the Coral running, but it hangs at the first frame. I think it's related to an unsupported configuration on my workstation. It's Ubuntu 20.04 at the moment and the Python version I am installing is an unofficial build of 3.7 in a venv. I will try on one of my other x86 Linux boxes which are all at 18.04.

mdegans, 2020-05-02:
So, the problem was the setup. Coral works plugged into my server without an issue, and the performance is much better than with CPU version.

[image]

All that is left is Jetson support. I will put something temporary for the PR and integrate DeepStream shortly after, now that the groundwork is there for it.

A new version of DeepStream was released anwyay and I want to have a look at the changes before doing anything. I browsed the documentation this morning and it doesn't appear that there are any breaking changes, so code I already have can probably be adapted without major modification.

mdegans, 2020-05-03:
Almost done reintegrating the Jetson TensorRT code.

Do you have any objection to:
* Rename the Jetson backend to TensorRT, since it will probably work on any TensorRT platform including x86/Nvidia
* Rename the x86 backend to TensorFlow since it will probably work on any TensorFlow platform
* Add DeepStream/GStreamer as a separate backend rather than replacing the Jetson one?
 My thinking is that it's good to have fallbacks in case one backend isn't available or fails.

mdegans, 2020-05-03:
The TensorRT backend is working. Just need to update some Dockerfiles, test some things, and you'll have a PR. Sorry for the delay, but I got sidetracked. I'm ADD AF. Ooh, squirrel!
[image]

mhejrati, 2020-05-04:
> * Rename the Jetson backend to TensorRT, since it will probably work on any TensorRT platform including x86/Nvidia

This is a bit complicated since we package apps based on the host architecture. E.g. we may need a TensorRT release for ARM and another one for x86. As for the codebase, it is ok to use TensorRT instead of Jetson but our logic was to simplify it by segmenting codebase and container packaging in the same way, therefore we decided to go with the host types instead of SW architecture. Does it make sense? I am interested to hear your feedback.

mhejrati, 2020-05-04:
> * Add DeepStream/GStreamer as a separate backend rather than replacing the Jetson one?
>   My thinking is that it's good to have fallbacks in case one backend isn't available or fails.

It makes a lot of sense. Thanks.

mdegans, 2020-05-04:
Re: image names, docker hub supports multiple architectures for a given image name.  If you've noticed, for example:
```Dockerfile
FROM ubuntu:bionic
```
Will work on nearly any platform. You can take a look at the os/arch dropdown [here](https://hub.docker.com/layers/ubuntu/library/ubuntu/latest/images/sha256-2b90cad5ded7946db07a28252618b9c8b7f4f103fc39266bcc795719d1362d40?context=explore) and see what I mean.

I might suggest simply `neuralet/smart_distancing:$VERSION` with a `latest` alias to the latest version.

When the container is run, smart_distancing, picks the best solution for the platform it's running on. If x86 or aarch64, it can pick Nvidia, Coral, or Software fallbacks. This way there are fewer versions and it "just works". Thoughts?

mhejrati, 2020-05-04:
I agree that fewer versions and "just works" is preferred, but we haven't done a lot of testing on this yet. Would it be possible to not make this change and add it as an issue to get back to once we do a bit more design discussion and test? This hopefully unlocks you to get your PR in and give us enough time to discuss all these very valid suggestions.

mdegans, 2020-05-04:
That's no problem at all. I haven't actually made the proposed change so there is nothing to undo. Just let me run some final tests and I'll submit a PR so you can merge into a (dev?) branch.

mhejrati, 2020-05-04:
Great, I think it is ok to make the PR against the master branch, I can do some testing and review quickly and merge.

mdegans, 2020-05-04:
Ok. I haven't tested the Dockerfiles yet but I made some quick changes. They'll probably work fine even with the old Dockerfiles so long as the entrypoint is changed. LMK whether you want me to submit as-is.

mhejrati, 2020-05-04:
It is fine if you submit as-is, I can test on my end :)

mdegans, 2020-05-04:
Ok. Lemme create the PR then :)

mdegans, 2020-05-04:
Couple questions:
* do you have a preferred docstring format (eg. Google style)?
* do you want me to target any previous versions of JetPack or just the latest?

mhejrati, 2020-05-04:
- We are using PEP8 coding style, slowly enforcing it throughout the codebase
- Currently Jetpack 4.3, we had issues with 4.4 so it is better to stick to 4.3 for now while we test 4.4 more thoroughly. Does that work for you?

mdegans, 2020-05-04:
> * We are using PEP8 coding style, slowly enforcing it throughout the codebase

Gotcha. I mean the docstring style tho, not the code style. Documentation parsers like sphinx that automatically generate documentation based on docstrings, but there are different formats people use. Sphinx handles them all, but the resulting docs can look different depening on the input format. I don't really have a preference, but it's probably best for formatting sake that an agreed-upon style is used. A favorite, and used a lot in your codebase, seems to be [Google style](https://sphinxcontrib-napoleon.readthedocs.io/en/latest/example_google.html).

> * Currently Jetpack 4.3, we had issues with 4.4 so it is better to stick to 4.3 for now while we test 4.4 more thoroughly. Does that work for you?

That's no problem. I have 4.4 on my Xavier to test with and 4.3 on my Nano. I am working on getting it working on 4.4 first, but will backport the changes.

It

mdegans, 2020-05-04:
Hey, is the issue you are having with 4.4 related to libflattenconcat by chance?

mdegans, 2020-05-04:
I have a couple notes re: docker versions
* tegra nvidia-docker images are not forwards or backwards compatible becuase of it's approach of bind mounting libraries
* on 4.3, tensorrt's python bindings need to be installed in the image, but on 4.4 they are automatically bind mounted (see `/etc/nvidia-container-runtime/host-files-for-container.d/tensorrt.csv`)
* on 4.3. libflattenconcat's source is available as part of the libnvinfer-samples package under `/usr/src/tensorrt/samples/python/uff_ssd`. On 4.4 the source is not available. The README changelog has not been updated and does not say where it went. The examples have removed flattenconcat's usage entirely. Any clues on why?

JsonSadler, 2020-05-06:
I read this long conversation, and I think the PR made by @mdegans is taking this project to another level. I may review the changes before I fall asleep :D

mhejrati, 2020-05-06:
@mdegans Thanks for the PR, I am reviewing it, sorry for the radio silence, I will add comments later tonight.

mdegans, 2020-05-06:
@mhejrati 

No rush. I'm writing tests for what I wrote yesterday (not comitted yet). I expect that'll take me the rest of the day (at least, depending on how smoothly it goes).

mhejrati, 2020-05-08:
@mdegans I reviewed your PR and commented on it, as discussed it is merged into branch deepstream-gstreamer.

mdegans, 2020-05-08:
I saw. Ty. I finished some tests for the basic GStreamer class I added the other day, but I am running into some trouble with the DeepStream components. Nothing major, just a bit of a pain.

Issue is that the Nvidia inference element absolutely demands a config file for some things (eg. model filenames). Setting these values directly via GObject properties is unsupported at the moment.

If you take a look at the GstEngine class, the various element creation methods set properties on GStreamer elements based on keys and values from the suppled configuration.

My plan is to change the logic in the inference creation function on the DsConfig subclass to write out an .ini for the keys and values instead, and then simply set the config file path property. It's more code but I don't see a way around it and I doubt you want multiple config files.

I like your simple config where you just specify a model name and boom, it just works.

linux-devil, 2020-05-08:
I got some time this weekend , if you want me to contribute in some way , feel free to tell me

linux-devil, 2020-05-08:
@mdegans Never tried to set these properties via GObject properties , but I did for gst-dsexample , i had to make some application level changes. Trust you with your logic and approach , let me know if I can be of any help and I will be happy to explore

mhejrati, 2020-05-08:
@linux-devil have you checked the open issues? Any of them look interesting to you? You are also more than welcome to open an issue yourself as well. 

Right now the high priorities are getting the DeepStream working, but there is some other work either in the computer vision part and the DevOps part that can be very helpful. 

Please let me know what kind of contributions you prefer and I can share more ideas.

linux-devil, 2020-05-08:
I was working on deepstream for my recent projects , let me check on it and update

mdegans, 2020-05-08:
So, to update, the temporary config writing code works and the nvinfer element parses it, so the approach I mentioned above works. Also wrote documentation and tests. The DsEngine pipeline starts and stops as well as it's base class, GstEngine, which loads and plays video files from local and network sources. It's choking on the .avi, but I think that's a decoder issue. Online sources work fine, even multiple ones at once.

rough todo:
* DsConfig needs some parameters for the resolution, which can be set from the master config.
* end to end testing with a local display
* end to end testing with html5 video output
* port the distancing calculation code (should be straighforward since I'm using the same format).
* osd drawing code (this has to be done in the pipeline with the nvosd element, but I can recycle a lot of code).
* frontend html5 video template
* probably something I am forgetting

All in all, it's coming well, but it's taking longer than expected, so sorry about that. The time hasn't been wasted, however. It mostly all has documentation and tests, so if anybody else jumps in at any point it should be easy to work with. GStreamer isn't friendly, but I'm tying to hide it's complexity. I welcome any comments on the API for GstEngine, GstConfig and friends.

mhejrati, 2020-05-09:
> All in all, it's coming well, but it's taking longer than expected, so sorry about that. The time hasn't been wasted, however. It mostly all has documentation and tests, so if anybody else jumps in [...]

Your contribution here takes this project a huge step forward :) I am so hopeful that this can help in making it safer for people going back to their normal life! 

Is the code on your fork to review?

valdivj, 2020-05-09:
mdegans
 While attempting to run the Tests in the "deepstream-gsteamer"  fork I  noticed the paths in the test_XXX.py samples have the path:
SAMPLES_PATH = '/opt/nvidia/deepstream/deepstream/samples/'
When deepstream installs its path is like this:
SAMPLES_PATH = '/opt/nvidia/deepstream/deepstream-4.0/samples/'
Once I changed the path in the test_xxx.py samples
I was able to run the tests succesfully.

By the way to everybody working on this project  
I SALUTE YOU

rk68, 2020-05-09:
Salute!

JsonSadler, 2020-05-09:
> So, to update, the temporary config writing code works and the nvinfer element parses it, so the approach I mentioned above works. Also wrote documentation and tests. The DsEngine pipeline starts [...]

I am reading your fork, where you are implementing deepstreem under the detectors directory, would you please let me know why are you doing this as a separate detector? (I expected that detectors are separated according to the architecture). 

BTW, would you please give me a starting point? I am a little bit lost in the code and I want to help with the DeepStream branch of yours so we can make the PR together.

JsonSadler, 2020-05-09:
> So, to update, the temporary config writing code works and the nvinfer element parses it, so the approach I mentioned above works. Also wrote documentation and tests. The DsEngine pipeline starts [...]

Thanks for PR #64 to use gstreamer back-end of OpenCV, I will make another PR to stream the output with Gstreamer too, just wanted to let you know.

mdegans, 2020-05-09:
> review?

@mhejrati 

It's WIP, but you can take a look at the direction of the design. GstEngine is where most of the work is done. I plan on that engine being the guts of the DsDetector class (and future GStreamer based detectors), with the same interface used by the other Detectors. It's mostly all documented in the docstrings, which build to real documetation with examples.

Re: your suggested changes to the config and Detector class model loading, I plan on adding those soon. My next commit will be integrating the main config, so i'll probably do it at the same time or right after.

mdegans, 2020-05-09:
> While attempting to run the Tests in the "deepstream-gsteamer" fork I noticed the paths in the test_XXX.py samples have the path:
> SAMPLES_PATH = '/opt/nvidia/deepstream/deepstream/samples/'

Ty for spotting that. That's the path for DeepStream 5.0 (it's a symlink to `/opt/nvidia/deepstream/deepstream-5.0/`), but I need to make sure it works with both versions.

mdegans, 2020-05-09:
> I am reading your fork, where you are implementing deepstreem under the detectors directory, would you please let me know why are you doing this as a separate detector? (I expected that detectors are separated according to the architecture).

A DsEngine (internal to DsDetector) will work on any platform DeepStream supports, x86-64/Nvidia or aaarch64/Nvidia. Likewise there is no architecture specific code in GstEngine. When I make a CoralEngine, powered by GstEngine, it should run on any platform Coral supports (aarch64 or x86-64). It might even work on Windows/Coral without much trouble.

Those are the internal Engine classes however. Detector and Distancing classes can still have architecture specific restrictions if you choose. I don't plan on changing your design unless that's what y'all decide on.

The Engines classes exist only because GStreamer has it's own event loop and async code doesn't play well with sync code (or other event loops). The threading module was an option, but it's still bound by the GIL and it would hurt performance too much.

> BTW, would you please give me a starting point? I am a little bit lost in the code and I want to help with the DeepStream branch of yours so we can make the PR together.

`deepstream/_gst_engine.py` is a good place to start if you want a preview.

You will need DeepStream and the Python DeepStream bindings installed to actually do anything. `sudo apt install deepstream-4.0` (or 5.0, depending on your JetPack) will install DeepStream itself. Instructions for the Python bindings can be found [here](https://github.com/NVIDIA-AI-IOT/deepstream_python_apps/blob/master/README.md#python-bindings). Having pyds installed is a requirement to build the `smart_detection.detectors.deepstream` docs _for now_.

mdegans, 2020-05-09:
> Thanks for PR #64 to use gstreamer back-end of OpenCV, I will make another PR to stream the output with Gstreamer too, just wanted to let you know.

That would be very helpful. Let me know if you need any help with gstreamer pipeline strings. There are hundreds of elements, many doing the same thing but not all have the same level of polish. GStreamer elements are classified as "good", "bad" and "ugly". You can tell which group they are in, as well as their capabilites and design, with the `gst-inspect-1.0` command.

Useful would be getting a working sink working to serve up html5 video to the frontend. You might experiment with the elements: tcpserversink, [webrtc](https://opensource.com/article/19/1/gstreamer), and various muxers (eg. webm).

If you can find a working sink solution (the end of the pipeline) we can reuse that everywhere, both in the legacy code with OpenCV's underlying Gstreamer support, and inside the GstEngine based classes. I'd recommend using gst-launch-1.0 to rapidly test designs for the end of the pipeline.

mdegans, 2020-05-10:
So, to update, i'm reverting to using deepstream config files. There are just too many properties to set without it. For DeepStream, there will have to be a .ini file for each inference engine. Also, they're copyrighted under a restrictive license, as well as all derivatives, so they can't be packaged with smart_distancing.

Example:
```ini
# Copyright (c) 2018-2020 NVIDIA Corporation.  All rights reserved.
#
# NVIDIA Corporation and its licensors retain all intellectual property
# and proprietary rights in and to this software, related documentation
# and any modifications thereto.  Any use, reproduction, disclosure or
# distribution of this software and related documentation without an express
# license agreement from NVIDIA Corporation is strictly prohibited.

# Following properties are mandatory when engine files are not specified:
#   int8-calib-file(Only in INT8)
#   Caffemodel mandatory properties: model-file, proto-file, output-blob-names
#   UFF: uff-file, input-dims, uff-input-blob-name, output-blob-names
#   ONNX: onnx-file
#
# Mandatory properties for detectors:
#   num-detected-classes
#
# Optional properties for detectors:
#   cluster-mode(Default=Group Rectangles), interval(Primary mode only, Default=0)
#   custom-lib-path,
#   parse-bbox-func-name
#
# Mandatory properties for classifiers:
#   classifier-threshold, is-classifier
#
# Optional properties for classifiers:
#   classifier-async-mode(Secondary mode only, Default=false)
#
# Optional properties in secondary mode:
#   operate-on-gie-id(Default=0), operate-on-class-ids(Defaults to all classes),
#   input-object-min-width, input-object-min-height, input-object-max-width,
#   input-object-max-height
#
# Following properties are always recommended:
#   batch-size(Default=1)
#
# Other optional properties:
#   net-scale-factor(Default=1), network-mode(Default=0 i.e FP32),
#   model-color-format(Default=0 i.e. RGB) model-engine-file, labelfile-path,
#   mean-file, gie-unique-id(Default=0), offsets, process-mode (Default=1 i.e. primary),
#   custom-lib-path, network-mode(Default=0 i.e FP32)
#
# The values in the config file are overridden by values set through GObject
# properties.

[property]
gpu-id=0
net-scale-factor=0.0039215697906911373
model-file=../../models/Primary_Detector/resnet10.caffemodel
proto-file=../../models/Primary_Detector/resnet10.prototxt
model-engine-file=../../models/Primary_Detector/resnet10.caffemodel_b30_gpu0_int8.engine
labelfile-path=../../models/Primary_Detector/labels.txt
int8-calib-file=../../models/Primary_Detector/cal_trt.bin
batch-size=30
process-mode=1
model-color-format=0
## 0=FP32, 1=INT8, 2=FP16 mode
network-mode=1
num-detected-classes=4
interval=0
gie-unique-id=1
output-blob-names=conv2d_bbox;conv2d_cov/Sigmoid
force-implicit-batch-dim=1
#parse-bbox-func-name=NvDsInferParseCustomResnet
#custom-lib-path=/path/to/libnvdsparsebbox.so
## 0=Group Rectangles, 1=DBSCAN, 2=NMS, 3= DBSCAN+NMS Hybrid, 4 = None(No clustering)
#cluster-mode=1
#scaling-filter=0
#scaling-compute-hw=0

[class-attrs-all]
pre-cluster-threshold=0.2
group-threshold=1
## Set eps=0.7 and minBoxes for cluster-mode=1(DBSCAN)
eps=0.2
#minBoxes=3
roi-top-offset=0
roi-bottom-offset=0
detected-min-w=0
detected-min-h=0
detected-max-w=0
detected-max-h=0

## Per class configuration
#[class-attrs-2]
#threshold=0.6
#eps=0.5
#group-threshold=3
#roi-top-offset=20
#roi-bottom-offset=10
#detected-min-w=40
#detected-min-h=40
#detected-max-w=400
#detected-max-h=800
```

@mhejrati 

Can you contact me privately at michael.john.degans@gmail.com

There is an issue regarding this I currently can't share publicly.

mdegans, 2020-05-11:
So, I simplified the GstConfig class to just wrap the master config since I figured all the necessary options for the Gstreamer elements (eg. desired output resolution) can be gotten from the main config direcltly. Everything seems simpler this way.

```python
class GstConfig(object):
    """
    GstConfig is a simple class to wrap a ConfigEngine and provide
     for a GstEngine.

    Arguments:
        master_config:
            The master :obj:`ConfigEngine`_ to use internally.
    """
```
It also treats the ConfigEngine as a source of truth, so if it's modified, and the engine is restarted, the configuration can be updated without restarting the entire app (at least, that's the hope).

I will push these changes soon. I also:
* started writing the DeepStream implementation of Detector and Distancing.
* sketched out a new idea for quick calculation of danger

mdegans, 2020-05-11:
So, to update, it may not be possible to support deepstream 4.0 since Nvidia seems to have removed the download for the 4.0 python bindings. I have [asked on the forum](https://forums.developer.nvidia.com/t/how-to-obtain-the-old-4-0-python-bindings/122771) if there is a link to them. Awaiting a response. For licensing reasons, we can't include the 5.0 configs, so it may be necessary to use the ones installed on the system.

JsonSadler, 2020-05-12:
Okay seems things got complicated, what if we use a gpu accelerated version of gstreamer, write a plugin to do the inference and get the results back - while still in GPU's memory - draw them on the frame and/or send them over some streaming protocol as metadat with the frames. Thoughts?

[14 more replies]

closed 2022-12-12 by mhejrati