galliot-us/neuralet #55: Could you create a bird's eye view for surveillance video? I think it will be useful and I saw some location accuracy problems when I tried to map camera view to bird's eye view.

PhamDuyHieutb, 2020-05-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

PhamDuyHieutb, 2020-05-06:
I think it will be useful for your lib and I saw some location accuracy problems when I tried to map camera view to bird's eye view.

mhejrati, 2020-05-08:
@PhamDuyHieutb Can you please elaborate? There is currently a bird's eye map of detected persons but I imagine you are looking for something else.

PhamDuyHieutb, 2020-05-08:
I mean that in your smart social distancing app, you can add a bird's eye map as a feature in your open source. (for example in this video  https://www.youtube.com/watch?v=15iIV1Lff-M)
I think it will be useful for your project.

mhejrati, 2020-05-08:
It is already implemented, have you tried our latest version?

closed 2020-05-08 by mhejrati