galliot-us/neuralet #57: Increase code coverage

mhejrati, 2020-05-07:
Build unit tests and end-to-end tests for the smart social distancing application.

Looking for community help here @mrn-mln @emmaWdev

mdegans, 2020-05-08:
Some might find it easier to start with documemtation/tests and end with code. I try to do that when I can since it ends up saving time in the end.

Re: coverage in the deesptream branch, some things have simple doctess and GstEngine has simple end-to-end tests for the pipeline and local and network video sources. I haven't set up a .coveragerc yet, however.

mhejrati, 2020-05-10:
We can keep this issue around to brainstorm testing strategies, including setting up .coveragerc and more end-to-end testing as well. It is also connected to #56 and how we want to wire up a CI/CD for it.

closed 2022-12-12 by mhejrati