galliot-us/neuralet #63: Support Standard Input Sources for Smart Social Distancing

mhejrati, 2020-05-07:
We need to add support for the current input sources:

- Offline Video Files (either stored locally or hosted online)
- Live Video Feeds (IP Camera, etc)
- CSI on Jetson and Coral

mdegans, 2020-05-07:
How would you like the config file to look?

Right now we have 
```ini
[App]
VideoPath: /some/file.ext
...
```

I might suggest
```
VideoUri: rtsp://hostname/path
```
for uri sources and
```
VideoCamera: (camera path or id)
```
for video cameras

rk68, 2020-05-07:
@mdegans - could you send an email covid19project@neuralet.com? I want to discuss something with you offline.

mhejrati, 2020-05-08:
> I might suggest
> 
> ```
> VideoUri: rtsp://hostname/path
> ```
> 
> for uri sources and
> 
> ```
> VideoCamera: (camera path or id)
> ```

Makes sense.

mdegans, 2020-05-08:
See #64 for an idea of how this might be accomplished. Work will need to be done on the config parser class to make the suggested changes, and the camera support isn't in that commit, but one potential approach is in there there. I will focus on DeepStream for now, but that's a starting point for whoever wants to pick this up.

mhejrati, 2020-05-08:
Great, thanks @mdegans for putting together the suggestion. 
@linux-devil it will be great if you can help with this.

mdegans, 2020-05-11:
So, one issue with .ini files is that Sections can't have duplicate keys (at least without modifying the parser), so I just solved this in the deepstream branch by having Source sections. This is the same solution that deepstream_app takes with it's app level .ini files.

So, currently, my proposed deepstream.ini looks like this:

```ini
[App]
Host: 0.0.0.0
Port: 8000
Resolution: 640,480

[Source_0]
; VideoPath may be a uri supported by uridecodebin (rtsp, http, etc.) 
; or a local file.
; TODO(mdegans): camera sources.
VideoPath: /opt/nvidia/deepstream/deepstream-5.0/samples/streams/sample_720p.h264

[Source_1]
; VideoPath may be a uri supported by uridecodebin (rtsp, http, etc.) 
; or a local file.
; TODO(mdegans): camera sources.
VideoPath: /opt/nvidia/deepstream/deepstream-5.0/samples/streams/sample_720p.h264

[Source_3]
; VideoPath may be a uri supported by uridecodebin (rtsp, http, etc.) 
; or a local file.
; TODO(mdegans): camera sources.
VideoPath: /opt/nvidia/deepstream/deepstream-5.0/samples/streams/sample_720p.h264

[Detector]
; Supported devices: Deepstream
Device: DeepStream
; Detector's Name can be "resnet10" or "peoplenet"
Name: resnet10
;ImageSize is not needed since this is included in the deepstream config .ini
ClassID: 3
MinScore: 0.25

[PostProcessor]
MaxTrackFrame: 5
NMSThreshold: 0.98
; distance threshold for smart distancing in (cm)
DistThreshold: 150
; ditance mesurement method, CenterPointsDistance: compare center of pedestrian boxes together, FourCornerPointsDistance: compare four corresponding points of pedestrian boxes and get the minimum of them.
DistMethod: CenterPointsDistance

[Logger]
; options: csv, json (default is csv if not set)
Name: csv
; optional log path (default to ~/.smart_distancing/logs/):
;LogDirectory: /some/path
```

This would also allow per-source options to be set, not just the path. Thoughts?

mhejrati, 2020-05-13:
This makes sense to me.

mdegans, 2020-05-14:
So, multiple standard sources are implemented in the Gstreamer branch and I'm currently paused waiting on the pipeline end and distancing code. @mhejrati Any objection to me continuing where I left off with #64 so this same config format can be used elsewhere?

mhejrati, 2020-05-15:
No objections @mdegans

closed 2022-12-12 by mhejrati