galliot-us/neuralet #67: Output stream with Gstreamer

JsonSadler, 2020-05-09:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

JsonSadler, 2020-05-09:
@mhejrati Please assign this to me

JsonSadler, 2020-05-09:
@mdegans Just be informed that I am doing this

mdegans, 2020-05-09:
Sounds good. `cv2.VideoWriter` is a good way to accelerate things for the moment.

You may wish to refer to [this plugin guide](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/accelerated_gstreamer.html) for Tegra.

While decoding pipelines can be built automatically with these elements using uridecodebin, you may need to manually select encoders for an encoding pipeline. You can inspect the details of those elements with gst-inspect-1.0. There are _many_ examples in the above document.

Optimally, a pipeline should use NVMM memory from beginning to end, so, for example, if you need to scale the video, you can use nvvideoconvert or nvvidconv rather than videoconvert for best performance. After video is encoded it doesn't matter (there are no GPU accelerated network sinks, for example, becuase network code would probably run very poorly on a GPU).

mhejrati, 2020-05-10:
@JsonSadler Sounds good, thanks.

mdegans, 2020-05-12:
How's it coming @JsonSadler ?

Any good network sinks we can use? Normally, I use RTSP, but we're targeting web browsers, so you'll have to find some other sinks. Once you have a working solution for the end of the pipeline we can use the string anywhere (GStreamer static pipelines or bins can be described as a "string ! of ! elements").

you could use webmmux for a webm container. You need to encode to vp8/9 first. Not all platforms support each, so you'll have to pick the right encoder element for the right platform.

Then you can mux with `webmmux` with the `streamable=true` option. Finally, you need some way to serve up the stream. `multisocketsink` can be used for that. There's a blog entry with a linked example in C here:

https://coaxion.net/blog/2013/10/streaming-gstreamer-pipelines-via-http/

You can probably also use `tcpserversink` like in the following, but multisocketsink` seems more oriented towards multiple connections:

https://stackoverflow.com/questions/25810640/gstreamer-tcpserversink-v0-10-vs-1-0-and-html5-video-tag

So, you'll have to Google and experiment, but this might work:

`platform specific vp8 encoder here ! webmmux streamable=true ! tcpserversink host=127.0.0.1 port=8080`

And as the example suggests:
```html
    <video autoplay controls width=320 height=240>    
    <source src="http://localhost:8080" type="video/webm">
       You browser doesn't support element <code>video</code>.
    </video>
```

You could also use a h264 encoder (see the guide for nvidia's choices), `qtmux` which muxes mp4 containers, and change the video tag type to `video/mp4`. I'm not certain, but you [may also have to specify a codec explicitly](https://tools.ietf.org/html/rfc6381#section-3.6) if it doesn't work without it. mp4 / webm are just container formats and can contain different video / audio codecs.

Here's a link that might help you with that:
https://stackoverflow.com/questions/16363167/html5-video-tag-codecs-attribute

Anyway. LMK if you run into trouble. I'm not very experienced with web video (I'm hoping you can learn this part), but I know gstreamer well enough, so if you run into trouble, lmk.

mhejrati, 2020-05-13:
@JsonSadler @mdegans 
FYI: #78 is in progress, working on a new fast api backend and react frontend. 

Exciting to wire these all up ;)

mhsekhavat, 2020-05-14:
Hi everyone. I am going to continue implementing gstreamer. I have pushed a WIP commit in `gstreamer` branch that writes to `tcpserversink` as @mdegans has mentioned before:

https://github.com/neuralet/neuralet/blob/678e3e4dffe90ef4e85fdfaaeaecb6b6042f0242/applications/smart-distancing/libs/core.py#L91-L96

I have moved the bounding box visualization logic from `web_gui.py` to the `libs/core.py` so that the visualization be done in the generated gstreamer stream.

The Docker build script for this branch takes a lot of time to build gstreamer and the frontend, you can pull the built image from dockerhub:

```bash
git checkout gstreamer
cd applications/smart-distancing
docker run -it -p 8000:8000 -p 8080:8080 -v $PWD/../..:/repo mhsekhavat/neuralet-smart-distancing-x86
```

Hi @mdegans -- thank you so much for contributing. Do you have any idea why it doesn't load the video in the browser once the processing time per frame is increased? When I run the code with 
https://github.com/neuralet/neuralet/blob/678e3e4dffe90ef4e85fdfaaeaecb6b6042f0242/applications/smart-distancing/libs/core.py#L102
it doesn't process the frames and just writes the frame to output stream. The can be seen in the frontend (`localhost:8000`) with a delay of ~15 seconds (I don't know why it takes 15 seconds to load the html5 video?). However, if I switch the `DO_PROCESS` variable to `True`, the video is never loaded in the frontend. Any ideas?

mhsekhavat, 2020-05-14:
@mdegans I am also curious which parameters of `vp8enc` and `webmmux` we should focus on for improving the streaming performance. I am looking for some parameters like buffer length.

mdegans, 2020-05-14:
> I have moved the bounding box visualization logic from `web_gui.py` to the `libs/core.py` so that the visualization be done in the generated gstreamer stream.

Thanks. I did that on my own fork for the same reason.

> The Docker build script for this branch takes a lot of time to build gstreamer and the frontend.

On the branch I checked out, Dockerfile-x86 gets them from apt packages, but building OpenCV. Is this the Dockerfile you are building from?

It may not be necessary to build OpenCv if you're willing to use an older version. My understanding is that gstreamer support should be alrready built into the version in the Debian repos used by tensorflow/tensorflow (but I haven't tested this, it is on Ubuntu)

fyi, you may wish to commit the logs dir, otherwise:
```
 $ docker run -it -p 8000:8000 -p 8080:8080 -v $PWD:/repo mhsekhavat/neuralet-smart-distancing-x86
Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 13, in __init__
    self.engine = CvEngine(self.config)
  File "/repo/applications/smart-distancing/libs/core.py", line 22, in __init__
    self.logger = Logger(self.config)
  File "/repo/applications/smart-distancing/libs/loggers/loggers.py", line 22, in __init__
    self.logger = csv_processed_logger.Logger(self.config)
  File "/repo/applications/smart-distancing/libs/loggers/csv_processed_logger.py", line 29, in __init__
    os.mkdir(self.log_directory)
FileNotFoundError: [Errno 2] No such file or directory: '/repo/applications/smart-distancing/ui/static/data'
```
I'm running it right now but not getting any frames. I'll take a look at this and update. I will take a look at this and update here if I can figure out what's going on.

re: performance, there is a bin you can use called encodebin that automatically can choose the best encoder for the system it's running on based on a supplied profile. It has profiles for webm/vp8

iirc, vp8enc is a software encoder and might perform poorly on embedded systems.

mdegans, 2020-05-14:
i see why it's not loading. I'm running it on a separate host but the html source is:
```html
<source src="http://localhost:8080/?1589497587489" type="video/webm">
```
It'll have to point to hostname in order to work, and the real one, not the one docker assigns. Not sure how to fix that Hmm...

mdegans, 2020-05-14:
So, it's not loading on Linux. Gonna try some other platorms and browsers. I get the page but no video.

mdegans, 2020-05-15:
So, it works fine with Firefox and `DO_PROCESS = False`

I put a print statement here at the end of the `if DO_PROCESS`:
```python
                        line_thickness=3,
                    )
                    print(cv_image.shape)
                else:
```
The right image shape is printed out, but very slowly (like 1-2 fps). It may be the encoder isn't getting frames fast enough. I will look at vp8enc and see if there is a parameter that can fix this, but it may be something is wrong with the distancing code that it's taking up so much time.

mdegans, 2020-05-15:
So, I commented out the image writing and it's still going super slow. I don't think the issue is the way the decoder is configured.

mdegans, 2020-05-15:
I played around with a bunch of parameters and eventually got it working with `keyframe-max-dist=1` (keyframe every frame) but it's going as slow as the print statemets suggest with DO_PROCESS=true. I'm willing to say that it's something in the distancing code. It appears something is wrong with the bounding box position as well.

[image]

mdegans, 2020-05-15:
I uploaded a video here so you can see what's going on with the bounding boxes:
https://youtu.be/mJSnIjwzvVY

If that problem is fixed, I think the current pipeline you have may work. I have no idea why it's not working with chrome, but i'm experimenting with this same thing on my NValhalla project, so if I figure out what the issue is there, I will share it.

I know somebody closely who works on Chrome, so I will ask if there is recommended way to do this other than putting a uri in the video tag.

mhsekhavat, 2020-05-15:
Hey @mdegans, great feedbacks!

> On the branch I checked out, Dockerfile-x86 gets them from apt packages, but building OpenCV. Is this the Dockerfile you are building from?

Yeah we build opencv and get the rest of packages from apt and pip. That's because the prebuilt opencv apt package of tensorflow image didn't support gstremer.

> fyi, you may wish to commit the logs dir, otherwise:
> ```
>  $ docker run -it -p 8000:8000 -p 8080:8080 -v $PWD:/repo mhsekhavat/neuralet-smart-distancing-x86
> Traceback (most recent call last):
>   File [...]

Will fix it. Thanks!

> re: performance, there is a ~bin you can use called encodebin~ (nevermind, it can't be used in a string pipeline) that automatically can choose the best encoder for the system it's running on based on a supplied profile. It has profiles for webm/vp8
> 
> iirc, vp8enc is a software encoder and might perform poorly on embedded systems.

Good points. I created #91 for further discussion.

> i see why it's not loading. I'm running it on a separate host but the html source is:
> 
> ```
> <source src="http://localhost:8080/?1589497587489" type="video/webm">
> ```
> 

Ah, I will make it configurable soon.

> So, it's not loading on Linux. Gonna try some other platorms and browsers. I get the page but no video. For comparison, what platform/browser are you using @mhsekhavat ?

I use Firefox. Didn't know it doesn't work with chrome!

> I uploaded a video here so you can see what's going on with the bounding boxes:
> https://youtu.be/mJSnIjwzvVY

Sorry I forgot to mention that I used a Dummy engine in this experimental branch. It detects random bounding boxes for the performance reason:
https://github.com/neuralet/neuralet/blob/678e3e4dffe90ef4e85fdfaaeaecb6b6042f0242/applications/smart-distancing/config-x86.ini#L9

> I know somebody closely who works on Chrome, so I will ask if there is recommended way to do this other than putting a uri in the video tag.

Great. Please let me know what you got.

Cheers!

mhsekhavat, 2020-05-15:
> I played around with a bunch of parameters and eventually got it working with `keyframe-max-dist=1` (keyframe every frame)

Good point. I will work more on it today.

mhsekhavat, 2020-05-15:
@mdegans Can you please provide some snippets on how to encode the bounding box meta data into frames and how the browser js code can extract and render them?

> Also it is hard to actually stream the data to the client side and visualize there if you are not using any external streaming protocol.

mdegans, 2020-05-15:
> > I know somebody closely who works on Chrome, so I will ask if there is recommended way to do this other than putting a uri in the video tag.

So, he says its stupidly hard to do web video and it's not implemented the way he would like, but he pointed me to [this](https://developers.google.com/web/fundamentals/media/mse/basics) which may (or may not) be of use.

On that page, something that caught my eye was:

> Some user agents, _Chrome for example, require a codec for mime types_ that don't self-describe their codecs. Rather than trying to sort all this out, it's better to just include both.

So, I think the source tag needs to include the codec type (vp8/vp9) in addition to the webm container type:

```
<source src="http://localhost:8080/?1589497587489" type='video/webm; codecs="vp8"'>
```

My guess that's why it works in FF but not in chrome.

We'll have to change the mime type depending on what source we're using since, for example, tx2 and Xavier support vp9 hardware encode while the Nano/tx1 does not (only vp8). Likewise I haven't been able to get Nvidia's hardware vp8 encoder working on my x86/gtx1080 box, so we might need to use `type='video/mp4; codecs="avc..."'` for some things. If we use mp4 and h274 I think we may also have to declare the profile with codecs, and there are many profiles, so that'll need some research.

It seems like tcpserversink may work for our purposes with these changes at least temporarily, but other solutions may be worth investigating like WebRTC. If we do move everything to GStreamer eventually, WebRTC may be worth looking at. There are demos here:

https://github.com/centricular/gstwebrtc-demos/
which i found linked off this blog entry:
https://opensource.com/article/19/1/gstreamer

Issue with that is I don't think we can as easily implement it with the OpenCV based code, so this would be if we move entirely to GStreamer in the future, with inference done in the same style as DeepStream or [RidgeRun's plugins](https://developer.ridgerun.com/wiki/index.php?title=GStreamer_DispTEC_Tracker_Plugin). I will experiment with it in my own project as it's written entirely in C and Vala (GObject C) and is more appropriate for this.

mhsekhavat, 2020-05-16:
Thank you @mdegans!
I have pushed some fixes to `gstreamer` branch.
The problems in your previous feedback are fixed. It works almost smoothly with `Dockerfile-x86-openvino`. However, it is still not working in chrome (even with type='video/webm; codecs="vp8.0, vorbis"') and needs further optimizations for other dockerfiles. 

I will switch to other tasks for now.

mdegans, 2020-05-16:
@mhsekhavat 

Did you try without vorbis, since there is no audio stream? It's just a guess, but it may be expecting it and not finding it. I will ask my Chrome friend if there is a way to get chrome to spit out some useful debug information about this and update.

mhsekhavat, 2020-05-16:
@mdegans yeah I tried with `'video/webm; codecs="vp8"` too. Thanks.

mdegans, 2020-05-16:
@mhsekhavat 

Ok. I'm going to try experimenting with gst-launch and see if I can get it working with Chome. My Chrome friend is busy at the moment but I will ask him about debugging this as soon as he's free.

mdegans, 2020-05-16:
So this works on Firefox on Xavier (on nano it would have to be `nvv4l2vp9enc` instead) :
```bash
gst-launch-1.0 filesrc location=./TownCentreXVID.avi ! decodebin ! nvvideoconvert ! nvv4l2vp9enc ! webmmux streamable=true ! tcpserversink host="0.0.0.0" port=8080
```
This is my test html:
```html
<!DOCTYPE html>
<html>
  <head>
    <title>NValhalla Web</title>
  </head>
  <body>
    <video width="640" height="480" controls>
      <source src="http://motherbird:8080" 
        type='video/mp4'/>
      <source src="http://motherbird:8080" 
        type='video/webm;codecs="vp9"'/>
      <source src="http://motherbird:8080"
        type='video/webm;codecs="vp8"'/>
    </video>
  </body>
</html>

```
I can't connect from Chrome or Firefox with this, but it appears to be serving up h264.
```
 gst-launch-1.0 filesrc location=./TownCentreXVID.avi ! decodebin ! nvvideoconvert ! nvv4l2h264enc ! h264parse ! qtmux streamable=true fragment-duration=10 ! tcpserversink host="0.0.0.0" port=8080
```
I will experiment more and update if I get something working with h264, or if I get the webm working with Chrome.

mdegans, 2020-05-16:
I got h264 working with firefox. still no chrome 
```
gst-launch-1.0 filesrc location=./TownCentreXVID.avi ! decodebin ! nvvideoconvert ! nvv4l2h264enc ! mpegtsmux ! tcpserversink host="0.0.0.0" port=8080
```
I found that solution [here ](https://stackoverflow.com/questions/25810640/gstreamer-tcpserversink-v0-10-vs-1-0-and-html5-video-tag)which says it should work with Chrome but not with FF. Oddly it's the other way round for me, and I'm not sure if that's the right muxer (it's usually for making .ts transport streams for blu ray and dvd and such).

mdegans, 2020-05-17:
So, Chrome friend pointed me to the dev console on Chrome and it's "invalid http response". So I guess we can't use plain tcp. Needs to be http. Makes sense. Googled and found [this](https://superuser.com/questions/443411/http-streaming-with-gst-launch-gstreamer). Going to look at hlssink next.

mdegans, 2020-05-17:
So, browsers don't support mpeg transport streams and need a container like mp4. Apparently [remuxing in .js is necessary if hlssink is used](https://stackoverflow.com/questions/60196390/gstreamer-and-hlssink-to-display-over-http-doesnt-work).

mdegans, 2020-05-17:
I found out [why firefox is working](https://gist.github.com/tetkuz/0c038321d05586841897#gistcomment-2727423).

> The reason for chrome throwing `ERR_INVALID_HTTP_RESPONSE` lies with tcpserversink not implementing HTTP. I.e. the reply on Port 8080 is only the payload/file, akin to transferring files with `netcat.` Chrome expects to see at least a bare header `HTTP/1.0 200 OK`, and refuses the stream.
> This appears not to be something that could be fixed with gst-launch-1.0.
> Meanwhile **Firefox (and likely other browsers as as well) simply deal with it ;-)**

So, it looks like hlssink and remuxing in .js is an option. I found this page with a working example linked off using gstreamer and nginx.

https://isrv.pw/html5-live-streaming-with-mpeg-dash

Problem there is we'd have to use an nginx instance. It might be a good idea anyways, but turns a single Dockerfile into a docker compose.

mhejrati, 2020-05-17:
Interesting!

We can do docker compose, it may be a sensible option as long as we can fit
it all on a single jetson or coral.

Also not supporting chrome is a totally valid solution too :)

mdegans, 2020-05-17:
> Interesting! We can do docker compose, it may be a sensible option as long as we can fit it all on a single jetson or coral. Also not supporting chrome is a totally valid solution too :)

@mhejrati 

Re: chrome. Every browser is Chrome now, tho. We kind of have to support it if we want the website to work on mobile browsers and such. Re: nginx and compose. I'm not opposed to that either and I've used nginx before for video streaming (forwarding from a GoPro). It went much smoother than this, so it might be a better option. It's an excellent web server. I will experiment with this on monday.

mdegans, 2020-05-18:
So, to update on compose, the issue now is that compose still doesn't support gpus [unless in swarm](https://github.com/docker/compose/issues/6691#issuecomment-526647914). I tried to hack around that with --device but it looks like that won't work anymore. 'runtime' might still work on Tegra, but [won't on x86](https://github.com/docker/compose/issues/6691#issuecomment-514429646), so compose's runtime option is out as well. Looks like using nginx (or docker compose at all) is out for now, but we might still be able to use hlssink if we get the existing frontend and server we have to point to the files.

What hlssink does is generate some video files and a playlist (m3u8) that points to them, so the video tag ends up looking something like:
```html
<video src="http://domain.com/playlist.m3u8">
```
But I'm reading this only works on Safari (and old Edge) out of the box becuase it's an Apple standard. On other platforms it's necessary to use a javascript library [like this](https://github.com/dailymotion/hls.js#getting-started), which sound like the only option. So at this point if we go with hls or webrtc we still need some javascript player to manage the video tag for us.

mhsekhavat, 2020-05-19:
@mdegans HLS looks interesting. I am gonna test it on x86 with the frontend. In the meanwhile, can you please let me know what encoder I should use for each of our current architectures?

mhsekhavat, 2020-05-19:
@mdegans, I could not serve the `m3u8` file in the server. Maybe you know what the problem is with:
https://github.com/neuralet/neuralet/blob/f9a129aabcc30696f786924aaf0bb880a8ef29ce/applications/smart-distancing/config-x86-openvino.ini#L8

For frontend, you can either use a raw html file using `hls.js`, or change 
https://github.com/neuralet/neuralet/blob/f9a129aabcc30696f786924aaf0bb880a8ef29ce/applications/smart-distancing/ui/web_gui.py#L63
in `gstreamer` branch to change the url for video. The frontend in `gstreamer` branch does include `hls.js`.

> It'll have to point to hostname in order to work, and the real one, not the one docker assigns. Not sure how to fix that Hmm...
It is now configurable in:
https://github.com/neuralet/neuralet/blob/f9a129aabcc30696f786924aaf0bb880a8ef29ce/applications/smart-distancing/config-x86-openvino.ini#L4

mdegans, 2020-05-19:
> @mdegans HLS looks interesting. I am gonna test it on x86 with the frontend. In the meanwhile, can you please let me know what encoder I should use for each of our current architectures?

For Tegra, the encoders are:

h264: nvv4l2h264enc
h265: nvv4l2h265enc
vp8 (nano, tx1, tx2 only): nvv4l2vp8enc
vp9 (tx2, Xavier only): nvv4l2vp9enc

For x86:
h264: vaapih264enc (hardware) x264enc (quality)
vp8: vaapivp8enc (hardware) vp8enc (software)
vp9: vaapivp9enc (hardware) vp9enc (software)

more vaapi encoders/decoders are available [here](https://gstreamer.freedesktop.org/documentation/vaapi/index.html?gi-language=c). There is a jpeg encoder as well, so there's always that as a fallback.

For edgetpu:
It looks like [google uses x264enc](https://coral.googlesource.com/edgetpuvision/+/refs/heads/master/edgetpuvision/pipelines.py#127) with the ultrafast preset. You can check out the parameters there. For decoding they choose decodebin (note: uridecodebin contains decodebin).

I just saw it does web streaming now, but what they've actually done is [written their own server](https://coral.googlesource.com/edgetpuvision/+/refs/heads/master/edgetpuvision/streaming/server.py) to serve up the video and they use the broadway javascript player to handle that on the frontend. We could use it since our license is identical, but might be more work to integrate into what's already there. It's usage seems [pretty nice](https://coral.googlesource.com/edgetpuvision/+/refs/heads/master/edgetpuvision/apps.py#27), however. The quality of code in that project is outstanding.

mdegans, 2020-05-19:
> @mdegans, I could not serve the `m3u8` file in the server. Maybe you know what the problem is with:
> https://github.com/neuralet/neuralet/blob/f9a129aabcc30696f786924aaf0bb880a8ef29ce/applications/smart-distancing/config-x86-openvino.ini#L8

Does it not generate the playlist? Is /tmp/hlssink within the web root? My mistake the other day was assuming it would serve it up itself. It doesn't. It requires you have an existing webserver and you just point `location` to the location within your web root you want to put the files, `playlist-location` to the name of the playlist, and `playlist-root` to the uri the `location` would evaluate to _once served up_. I will pull your changes and check it out.

mdegans, 2020-05-19:
I think i see the problem:
```python
        out = cv.VideoWriter(
            'appsrc ! videoconvert ! '
            f'{encoder}',
```
to join fstrings, both lines need to be prefixed with f

mdegans, 2020-05-19:
@mhsekhavat I submitted a PR for this. I also moved the videoconver element to the config file since Tegra has it's own element for that.

JsonSadler, 2020-05-21:
@mhsekhavat  @mdegans Great work guys! I was away for a long time, seems you have done what I was planning to do. Thank you @mhsekhavat

grdsvt, 2020-05-26:
Hi, I used Gstreamer and  cam to capture a video in real time.....
but I haven't good results 
I have 5-7fps but the indicated fps is 20.
Can someone help me to have best results?
Thanks

memina, 2020-06-23:
Hi, First of all, congratulations for your successful works.

I encountered the same problem on jetson nano.
Despite writing 28fps on flask, it shows very slowly around 2-3 fps.

How can I turn off the flask feature and see it on the screen with the cv.imshow command ?

JsonSadler, 2020-07-07:
@grdsvt @memina Master branch for [https://github.com/neuralet/smart-social-distancing](https://github.com/neuralet/smart-social-distancing) now streams the output with gstreamer, you can test that.

JsonSadler, 2020-07-07:
@memina @mdegans @mhsekhavat As output is now being streamed by gstreamer in branch master of the new repository, I suggest closing this issue, whilst it contains a huge amount of information that can be used in improving the solution later.

mdegans, 2020-07-07:
@JsonSadler 
Yeah, this milestone is reached. I would suggest closing out everything you can and pointing people to the new repo, but I think you're already on that.

closed 2020-07-07 by mdegans