galliot-us/neuralet #72: libnvinfer.so.6 no such file or directory

loophole64, 2020-05-11:
I'm setting up the smart distancing application on the Jeson Nano. I am using the sd card image from:
https://developer.nvidia.com/jetson-nano-sd-card-image

I have docker 18.09.7 installed.

I am following the instructions from:
https://galliot.us/2020/04/16/smart-distancing-technical-document/

I was able to build the docker image ok. When I try to run the docker container, I get the following error:

`Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 13, in __init__
    self.engine = CvEngine(self.config)
  File "/repo/applications/smart-distancing/libs/core.py", line 23, in __init__
    self.detector = Detector(self.config)
  File "/repo/applications/smart-distancing/libs/detectors/jetson/detector.py", line 17, in __init__
    from . import mobilenet_ssd_v2
  File "/repo/applications/smart-distancing/libs/detectors/jetson/mobilenet_ssd_v2.py", line 3, in <module>
    import tensorrt as trt
  File "/usr/local/lib/python3.6/dist-packages/tensorrt/__init__.py", line 1, in <module>
    from .tensorrt import *
**ImportError: libnvinfer.so.6: cannot open shared object file: No such file or directory**
`

I tried to check nvinfer version with dpkg -l | grep nvinfer and it shows version 7.1.0-1+cuda10.2. Does this mean I have the wrong version of the library?

I am a Linux rookie.

mhejrati, 2020-05-11:
Which Jetpack version are you using?

loophole64, 2020-05-11:
Correct me if I am wrong, but my understanding is that Jetpack is the software used on a host machine to flash the Jetson. I did not use Jetpack to flash the Nano. I downloaded an image for the SD card, which I referenced above.

Do I need to specifically get Jetpack 4.3 and re-flash the Jetson?

mhejrati, 2020-05-11:
The error could be because you haven't installed TensorRT. You can definitely install TensorRT without flashing the nano using Jetpack, see: https://docs.nvidia.com/deeplearning/sdk/tensorrt-install-guide/index.html

However, if you flash the SD card using JetPack it will come with all the packages required, including TensorRT. We have tested the app on JetPack 4.3 already and are working on 4.4.

loophole64, 2020-05-11:
TensorRT (v7.1.0-1) is installed. The image is from nVidia's Jetpack page, and they include everything Jetpack would include while flashing, I think. Maybe my version is too new. I will try flashing with Jetpack 4.3 when my barrel connector power supply gets here. Maybe I will try downgrading my version of TensorRT from 7 to 6 if I can figure that out. Thanks!

mdegans, 2020-05-11:
Note: JetPack and Docker base images actually have to match. Unfortunately, the way Nvidia-docker is implemented on Tegra, libraries like nvinfer are bind mounted at runtime rather than baked into the image, and even if they are included in the image, they will be mounted over by `--runtime nvidia`.

mdegans, 2020-05-11:
> try downgrading my version of TensorRT from 7 to 6 if I can figure that out.
I don't think that's possible since the package repositories only have one version.
Eg: the nvidia apt sources list for jetpack 4.4 is:
```
deb https://repo.download.nvidia.com/jetson/common r32.4 main
```
and 4.3 is:
```
deb https://repo.download.nvidia.com/jetson/common r32 main
```
Becuase they're different sources entirely, it's not possible to pin a version like you usually can with apt-get. Hopefully they change this approach in the future to use a single repo.

You could try an old debian package manually but there would probably be dependency issues.

JsonSadler, 2020-05-12:
> I'm setting up the smart distancing application on the Jeson Nano. I am using the sd card image from:
> https://developer.nvidia.com/jetson-nano-sd-card-image
> 
> I have docker 18.09.7 installed.
> 
> I am following the instructions from:
> https://galliot.us/2020/04/16/smart-distancing-technical-document/
> 
> I was able to build the docker image ok. When I try to run the docker container, I get the following error:
> 
> Traceback (most recent call last):
>   File "neuralet-distancing.py", line 22, in 
>     DistanceApp(args)
>   File "neuralet-distancing.py", line 13, in __init__
>     self.engine = CvEngine(self.config)
>   File "/repo/applications/smart-distancing/libs/core.py", line 23, in __init__
>     self.detector = Detector(self.config)
>   File "/repo/applications/smart-distancing/libs/detectors/jetson/detector.py", line 17, in __init__
>     from . import mobilenet_ssd_v2
>   File "/repo/applications/smart-distancing/libs/detectors/jetson/mobilenet_ssd_v2.py", line 3, in 
>     import tensorrt as trt
>   File "/usr/local/lib/python3.6/dist-packages/tensorrt/__init__.py", line 1, in 
>     from .tensorrt import *
> ImportError: libnvinfer.so.6: cannot open shared object file: No such file or directory
> 
> I tried to check nvinfer version with dpkg -l | grep nvinfer and it shows version 7.1.0-1+cuda10.2. Does this mean I have the wrong version of the library?
> 
> I am a Linux rookie.

From the instructions: 

> To run this application you require to set up your Jetson Nano using JetPack 4.3 and have Nvidia-docker runtime available.

Your Jetpack on the jetson Nano seems to be 4.4 if you directly got it from https://developer.nvidia.com/jetson-nano-sd-card-image. Instead, you have to flash 4.3 from here: [https://developer.nvidia.com/jetpack-43-archive](https://developer.nvidia.com/jetpack-43-archive)

As @mhejrati and @mdegans wrote before, nvidia docker works with mounting host's libnvinfer, etc. to the docker to get it work with the host's GPU, and as our docker images are built compatible with 4.3 ( `FROM nvcr.io/nvidia/l4t-base:r32.3.1` in Dockerfile for jetson) your host has to have 4.3 too. Changing this will break the code as it uses a tensorrt plugin built with and linked against TensorRT 6 which is available in Jetpack 4.3. We will solve this soon with building that plugin on the board itself and uploading docker images for both 4.3 and 4.4 so the user can easily push the proper image and the plugin will be built accordingly, but the implemntation is a little tricky and I do not have any idea when it can be released. I will let you know. 

For now, just try flashing 4.3 and it will work.

mdegans, 2020-05-12:
@JsonSadler 

Re: libflattenconcat.so

The plugin source is not available for TensorRT 7.0

It's ordinarily at `/usr/src/tensorrt/samples/python/uff_ssd/` but it's been removed in the latest release. I can only assume there is a reason for it. 

My recommendation for 4.4 Docker would be to use a deepstream base image from Nvidia. Finishing it maybe less work than getting the existing TensorRT code to work. It's up to you.

mhejrati, 2020-05-12:
@loophole64 is your issue resolved by flashing the board using JetPack 4.3? We are working on porting core CV engine to DeepStream and Gstreamer and supporting JetPack 4.4 you can track it here #50 

Until then, the easiest route is to use JetPack 4.3!

loophole64, 2020-05-12:
@mhejrati I was able to write the image from Jetpack 4.3 and resolve the issue. I had assumed the version I had was created was from 4.3 because it was under a section called "Jetpack 4.3" on nVidia's site. Thank you for your help!

mdegans, 2020-05-12:
> @mhejrati I was able to write the image from Jetpack 4.3 and resolve the issue. I had assumed the version I had was created was from 4.3 because it was under a section called "Jetpack 4.3" on nVidia's site. Thank you for your help!

If the version reported is incorrect, you may wish to report this to Nvidia so other avoid the same confusion. This sometimes happen, but Nvidia responds quickly to feedback on their forum.

mhejrati, 2020-05-13:
Awesome @loophole64. Closing this issue for now. Let us know if you have any other issues.

closed 2020-05-13 by mhejrati

fiv21, 2020-06-16:
Hi!
I'm Franco, from Argentina. I bought an AGX Xavier a few days ago and I've found this project extremely interesting, I'm wondering if there's any way I can help to you using my hardware and probably my knowledge to make some tests and support the new JetPack 4.4
Be in touch, all the best,

Franco.

rkgits, 2020-06-17:
Hi Franco,
This is Reza from Neuralet Team. Welcome to the project! We appreciate your
interest. Please join our Slack team and initiate. your discussion. There
are some other team members who can involve you more in this project.
https://join.slack.com/t/neuralet/shared_invite/zt-ez7x601j-1NehEt5LZxkQFg99ElYXbw

-Reza