galliot-us/neuralet #74: Could not load dynamic library and Memory Issue for Smart distancing

pritigavali, 2020-05-12:
Hi, 
I am following the instructions from:
[https://github.com/neuralet/neuralet/tree/master/applications/smart-distancing](https://github.com/neuralet/neuralet/tree/master/applications/smart-distancing)

My Machine Configuration - 
OS- Ubuntu 18.04
GPU- Nvidia GeForce 940M

I was able to build the docker image . When I try to run the docker container, I get the following error:

`2020-05-12 09:03:45.421680: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer.so.6'; dlerror: libnvinfer.so.6: cannot open shared object file: No such file or directory
2020-05-12 09:03:45.431507: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libnvinfer_plugin.so.6'; dlerror: libnvinfer_plugin.so.6: cannot open shared object file: No such file or directory
2020-05-12 09:03:45.431613: W tensorflow/compiler/tf2tensorrt/utils/py_utils.cc:30] Cannot dlopen some TensorRT libraries. If you would like to use Nvidia GPU with TensorRT, please make sure the missing libraries mentioned above are installed properly.
Downloading data from http://download.tensorflow.org/models/object_detection/ssd_mobilenet_v2_coco_2018_03_29.tar.gz
187932672/187925923 [==============================] - 341s 2us/step
2020-05-12 09:11:36.017416: W tensorflow/stream_executor/platform/default/dso_loader.cc:55] Could not load dynamic library 'libcuda.so.1'; dlerror: libcuda.so.1: cannot open shared object file: No such file or directory
2020-05-12 09:11:36.705277: E tensorflow/stream_executor/cuda/cuda_driver.cc:351] failed call to cuInit: UNKNOWN ERROR (303)
2020-05-12 09:11:37.051137: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:156] kernel driver does not appear to be running on this host (9d18664c1959): /proc/driver/nvidia/version does not exist
2020-05-12 09:11:38.717353: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2
2020-05-12 09:11:47.817744: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 1800000000 Hz
2020-05-12 09:11:49.181041: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x9e8f0e0 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2020-05-12 09:11:49.181396: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
Device is:  x86
Detector is:  mobilenet_ssd_v2
image size:  [300, 300, 3]
 * Serving Flask app "ui.web_gui" (lazy loading)
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: on
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
opened video  /repo/applications/smart-distancing/data/TownCentreXVID.avi
2020-05-12 09:16:07.029768: W tensorflow/core/framework/cpu_allocator_impl.cc:81] Allocation of 25159680 exceeds 10% of system memory.`

JsonSadler, 2020-05-12:
@pritigavali Hi!
I guess you are following the instructions for jetson nano (Dockerfile-jetson-nano), while you are using an x86 node. It is not documented yet, but supported if you clone the latest master branch. 

Just replace `Dockerfile-jetson-nano` with `Dockerfile-x86`. Let me know if it still does not run.

JsonSadler, 2020-05-12:
Oh my fault about not being documented, it is actually documented [Usage > Run on x86](https://github.com/neuralet/neuralet/tree/master/applications/smart-distancing#usage)

mhejrati, 2020-05-12:
@pritigavali is your issue resolved?

closed 2020-05-15 by mhejrati