galliot-us/neuralet #75: Add documentation for using with x86 (Dockerfile-x86) in Readme

JsonSadler, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
Yes, issue #74 happened.

**Describe the solution you'd like**
Add instructions for using with x86 (and desktop GPU)

**Describe alternatives you've considered**
-

**Additional context**
-

alpha-carinae29, 2020-05-12:
here is the instruction in `neuralet/neuralet/blob/master/applications/smart-distancing/README.md`
```
cd neuralet/applications/smart-distancing/

# 1) Build Docker image (This step is optional, you can skip it if you want to pull the container from neuralet dockerhub)
docker build -f Dockerfile-x86 -t "neuralet/x86_64:applications-smart-distancing" .
# 2) Run Docker container:
docker run -it -p HOST_PORT:8000 -v /PATH_TO_CLONED_REPO_ROOT/:/repo neuralet/x86_64:applications-smart-distancing
```

JsonSadler, 2020-05-12:
Oh nothing, it is already there, my fault.

closed 2020-05-12 by JsonSadler