galliot-us/neuralet #83: Deploying Social Distancing App

valdivj, 2020-05-13:
The final goal of the Social Distancing app is to deploy something can help business and people adhere to social distancing best practices.
Nuralet team, since this is your baby did you guys have any thought how this is going to be deployed to help the public.
Since I will be concentrating on the Jetson nano as my platform and as soon as the app is working smoothly on it. I was going to go to a few local business and ask the if I can deploy it at there locations.
So I can start getting real world feedback.

rkgits, 2020-05-13:
@valdivj - 
Thanks for opening this issue and working on it. 
This is the area that we did not make much progress but I am open to discuss this with you and see what is the best option to move forward. We can discuss further offline.

valdivj, 2020-05-13:
Reza

  I have noticed that Nvidia seems to have taken an interest in the success of you app.

https://news.developer.nvidia.com/jetson-project-of-the-month-smart-social-distancing-with-ai/

It seems that now is the right time and place for an app like this to flourish.

Like I mentioned in  my post as soon as I get my nano running the App with a frame rate of at least 24 fps I will take it some local business and try it out.

I want to see how accepting people are of being monitored. The plus side of this app is anybody can log onto to the streaming image and view it. This should show the customers that they are not being spied  on. And the business can boast that they are using the latest technology to keep there customers safe.

Does the app for the Nano have the heat map. I cant seem to activate it if it does. That feature is a must so it can show the hot spots.

Im am not an employee of Nvidia but I do have a sympathetic ear there.

You can located me on the Nvidia dev forum as @adventuredaisy

Here is also my you tube page showing what I have been doing with the Jetson nano;https://www.youtube.com/channel/UCOOidQm3y9w22O9n8yo30yQ

[https://yt3.ggpht.com/a/AATXAJwpGztRrk29yw8g9eYHf33X1vpYX1GD7Liw0w=s900-c-k-c0xffffffff-no-rj-mo]<https://www.youtube.com/channel/UCOOidQm3y9w22O9n8yo30yQ>

joev valdivia - YouTube<https://www.youtube.com/channel/UCOOidQm3y9w22O9n8yo30yQ>
www.youtube.com

<https://news.developer.nvidia.com/jetson-project-of-the-month-smart-social-distancing-with-ai/>

<https://news.developer.nvidia.com/jetson-project-of-the-month-smart-social-distancing-with-ai/>

mdegans, 2020-05-13:
So, one potential option is [Pypi](https://pypi.org/project/mce/). Another is [docker hub](https://hub.docker.com/r/mdegans/nvalhalla). At least that's what i've done with my projects. Then deployment is just a `pip install` or `docker run` away (or `compose up` or whatever).

At least in the deepstream-gstreamer branch i've done the necessary python packaging work, but I haven't tried an upload to [test pypi](https://test.pypi.org/) yet.

A third option is to create flashable system images. This may be ideal for the end user. In this case it's just a case of flashing a sd card / usb stick, booting the device, and configuring it.

valdivj, 2020-05-13:
1 vote for Docker

mhejrati, 2020-05-13:
I also suggest we go with the docker route. It has many benefits that we all know of.

@valdivj great to see your work on Jetson 👍 Once we have the app working, maybe you should create a video tutorial for it ;)

codebugged, 2020-06-29:
Hi @valdivj @mhejrati @rkgits 
Thank you for this repo.
How can I take live camera feed ( I am using Logitech C270 webcam)
Please help
Thanks in advance

mhaboali, 2020-06-30:
Hi all, 

I have the same question, how can I run the system with live camera input on Jetson Nano?

Thanks!

mats-claassen, 2020-06-30:
Hi @codebugged and @mhaboali, I think the question is a bit off topic in this thread, but have you tried replacing the path to the local video in the config file you are using, with either a link to the camera if it is connected over network, or the index of the webcam?

mhaboali, 2020-06-30:
Hi @mats-claassen thanks for your reply,

Yes I've tried that by using /dev/video0 in the config file. And I got "opened /dev/video0" in the terminal but I don't know how can I make sure it works properly. Could you help with that?

mats-claassen, 2020-06-30:
@mhaboali Have you joined our Slack channel? So we could continue the discussion there. Or, please open a different issue.

What do you mean by making sure it works properly? Have you opened the web dashboard? Are you seeing the camera feed there with bounding boxes for people? If so, then it is working.

JsonSadler, 2020-07-07:
One vote for docker and one vote for ready-to-flash images!
We are working on a new architecture which allows better deployment, I am going to merge changes @mdegans made long ago too, after changes made to the architecture, so we will have all three options to deploy it.

mhaboali, 2020-07-07:
Thanks @JsonSadler for keeping us updated!

I'm wondering now how can I run the system with PI camera V2 (CSI Camera)? I've tested the social distancing app with USB camera and it worked perfectly but the CSI camera I always get an issue with opening the camera.

JsonSadler, 2020-07-07:
> Thanks @JsonSadler for keeping us updated!
> 
> I'm wondering now how can I run the system with PI camera V2 (CSI Camera)? I've tested the social distancing app with USB camera and it worked perfectly but the CSI camera I always get an issue with opening the camera.

Would you please make a new issue for this so I can follow up? 
The smart-distancing-app is being extracted fromm current repo to another one, so please make your issue in the new repo here: [https://github.com/neuralet/smart-social-distancing](https://github.com/neuralet/smart-social-distancing)
Please first test with that new repo, then if you got any errors, post them there and describe your setup (Your board name, OS, etc.) so I can follow the problem step by step.

mhaboali, 2020-07-07:
Thanks, @JsonSadler for pointing me to the new repo, I highly appreciate your help.

Ok, I'll open the issue there shortly.

mhaboali, 2020-07-07:
Hi @JsonSadler 

I've tried it and I got issue with accessing the camera using gstreamer command inside the docker and I opened an issue there with all details https://github.com/neuralet/smart-social-distancing/issues/28

Thanks!

Candace-852, 2020-07-25:
[image]

the docker run command is giving this error , how to mitigate this?

emma-w-dev, 2020-07-25:
hi @Candace-852,
You need to replace the 'HOST_PORT' of running command with an open port on your host device.

JsonSadler, 2020-07-26:
Okay this issue went much off-topic, I suggest closing it and re-open the issue with more detailed concerns about deployment, in the new repository [here](https://github.com/neuralet/smart-social-distancing).

closed 2022-12-12 by mhejrati