galliot-us/neuralet #91: Smart Distancing: Tune gstreamer parameters for each architecture

mhsekhavat, 2020-05-15:
We should figure out the best encoder (alternative to `vp8enc`) for each platform. Maybe we can get some hints from how `encodebin` chooses the encoder.

mdegans, 2020-05-15:
Some notes that may be helpful to whoever takes this up:
* gstreamer has an internal rank for each element, and that's how encodebin chooses them along with the capabilities (caps)
* there is this code here that's one of the few examples using encodebin that i've found:
https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/-/tree/master/tests/examples/encoding
of interest, while reading the source, is [this section here](https://gitlab.freedesktop.org/gstreamer/gst-plugins-base/-/blob/master/tests/examples/encoding/gstcapslist.c#L122). Those code snippets are how you can get all elements of a minimum rank. The documentation is [here for c](https://gstreamer.freedesktop.org/documentation/gstreamer/gstelementfactory.html?gi-language=c#gst_element_factory_list_get_elements) and [here for python](https://lazka.github.io/pgi-docs/Gst-1.0/classes/ElementFactory.html#Gst.ElementFactory.list_get_elements).:

If you're unfamiliar with GObject C, `g_class_name_method_name(instance, param1, param2)` becomes `instance.method_name(param1, param2)` in python. Since C was not designed with object oriented programming in mind, this is the convention used.

There are [prettier languages](https://valadoc.org/gstreamer-1.0/Gst.ElementFactory.html) that wrap this ugliness and write the GObject C for you. There are examples in that documentation there for Vala. Vala is also really easy to bind for python ([bindings are "free"](https://github.com/antono/vala-object)). It's only real downside is it only plays well with GObject, but GStreamer is written in GObject, so it's perfect for that. If anybody chooses this admittedly eccentric language, lmk if you need any help (eg. with build files or cryptic error messages out of valac).

mdegans, 2020-05-15:
More notes on encoders:
* Nano/Tx1 do not support vp9 but do support vp8
* Tx2/Xavier does support vp9 but Xavier does not support vp8
* the plugin guide for Nvidia with some examples is [here](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/accelerated_gstreamer.html).
* the webm container format only supports vp8/vp9
* the mp4 conainer does not support vp8/vp9 but does support avc (h264) / hevc (h265)
* a support table of which container support which codecs is [here](https://en.wikipedia.org/wiki/Comparison_of_video_container_formats#Video_coding_formats_support).

mdegans, 2020-05-15:
Also, I found out that the Googler I know closely worked on the vp8 encoder and he said it's every effecient even in software (and I've seen it run on a potato), so if we have to have a software fallback, it's a good one.

closed 2022-12-12 by mhejrati