galliot-us/neuralet #95: Video file for face-mask detection

mrn-mln, 2020-05-16:
For testing face-mask detection, I need a video benchmark for creating a demo and test the capability of the model in a real-world scenario. Currently, we don't have video files in which people wear a face mask.

[image]

mrn-mln, 2020-05-18:
@rkgits, @mhejrati I've searched for the videos but it seems there are no proper videos for our task. However, there are lots of videos from news related to the face mask issue.
Sample videos:
https://www.youtube.com/watch?v=hekZBf8oUq0
https://www.youtube.com/watch?v=scStYflhltg
https://www.youtube.com/watch?v=HNMnu0Z7TgY
We can cut some parts of the videos and release a benchmark which is useful for the covid19 developer community.

mhejrati, 2020-05-20:
Some more videos with creative common license:
https://www.youtube.com/watch?v=MEj0ayTeEnU
https://www.youtube.com/watch?v=V54hhnyAntU
https://www.youtube.com/watch?v=5_0LobssR9A
https://www.youtube.com/watch?v=A77GS1Z5uvg
https://www.youtube.com/watch?v=EeuneefPTyY
https://www.youtube.com/watch?v=KM5oNfLlhnc
https://www.youtube.com/watch?v=WvK6QP8GqzY
https://www.youtube.com/watch?v=EIY9xJc4s0Q
https://www.youtube.com/watch?v=_FPFLmqYRC0
https://www.youtube.com/watch?v=ZAKOzzMZ2Bc
https://www.youtube.com/watch?v=UGX50EpWtPQ
https://www.youtube.com/watch?v=nf4bZgHsa5E
https://www.youtube.com/watch?v=HNMnu0Z7TgY

mrn-mln, 2020-06-03:
Good video for starting.
https://www.youtube.com/watch?v=7-IEW0cr5To
@mhejrati I also selected these videos which you've shared to start working on [issue 106](https://github.com/neuralet/neuralet/issues/106)
https://www.youtube.com/watch?v=V54hhnyAntU
https://www.youtube.com/watch?v=5_0LobssR9A
https://www.youtube.com/watch?v=EeuneefPTyY
https://www.youtube.com/watch?v=_FPFLmqYRC0

closed 2022-12-12 by mhejrati