galliot-us/neuralet #104: Generate TensorRT Engine from Face Mask detector model for Jetson

mhejrati, 2020-05-18:


emma-w-dev, 2020-05-18:
I'm working on a script and dockerfile to generate a TensorRT from a retrained ssd-mobilenet-v2. I will test and add ssd-mobilenet-v1 support to it for Face Mask model soon.

mhejrati, 2020-05-18:
Thanks @emma-w-dev

emma-w-dev, 2020-06-06:
hi,
I have commited the Dockerfile for TensorRT generation is [here](https://github.com/neuralet/neuralet/tree/master/training/tensorrt_generation/ssd),
and I have generated TensorRT from face-mask-detector [frozen graph from neuralet-models](https://media.githubusercontent.com/media/neuralet/neuralet-models/master/amd64/face_mask_detector_mobilenet_v1/mobilenet_face_mask_frozen_inference_graph.pb) and pushed it [here](https://github.com/neuralet/neuralet-models/blob/ea48b674ac5d93e2634a953d4b7f5b7e46906e7c/jetson-nano/face_mask_ssd_mobilenet_v1/TRT_face_mask_ssd_mobilenet_v1.bin).

closed 2022-12-12 by mhejrati