galliot-us/neuralet #105: Generate edge-TPU compiled TF-lite from Face Mask detector model for Coral

mhejrati, 2020-05-18:


mrn-mln, 2020-05-24:
@mhejrati I generated tflite model and sent a[ PR for this task](https://github.com/neuralet/neuralet-models/pull/5) on neuralet-models repo.

mhejrati, 2020-06-03:
Cool, thanks. 

Can you please provide more details of what you have done there and what are the next steps?

mrn-mln, 2020-06-03:
@mhejrati
A ssd-mobilenet-v1 detector is trained on face-mask dataset. I converted the model to quantized int8 model which is compatible with edgetpu devices (the mAP = 0.67%).
At the next step, I am going to use the tflite model for implementing a face/face-mask detector app at the social-distancing repo. The task is related to [issue 106](https://github.com/neuralet/neuralet/issues/106).

mrn-mln, 2020-06-09:
I tested a ssd-movilenet-v1, the generalization of the model was not well. So, I will try to train a new model for this task.
I will update my activity here.

mrn-mln, 2020-06-12:
I trained resnet-50 on WIDER FACE dataset. The mAP was ~0.52 which was not well enough. I will try to train a model on the HollywoodHead detection dataset in order to detect head.

closed 2022-12-12 by mhejrati