galliot-us/neuralet #110: I have problem to download model for jetson-nano

grdsvt, 2020-05-19:
Hi, I used :

sudo docker pull neuralet/jetson-nano

to  download model for Jetson-nano for social distancing, but I received the message:

Using default tag: latest
Error response from daemon: manifest for neuralet/jetson-nano:latest not found: manifest unknown: manifest unknown

Where is the error?
I see that is not present model for jetson too but in mail I received that it is available.
Can sameone help me?

Thanks

alpha-carinae29, 2020-05-19:
Hi
You can pull the Jetson Nano docker by running:
`docker pull neuralet/jetson-nano:applications-smart-distancing`
or build the container locally by clone this repository and run followings:
```
cd neuralet/applications/smart-distancing/

# 1) Download TensorRT engine file built with JetPack 4.3:
./download_jetson_trt.sh

# 2) Build Docker image (This step is optional, you can skip it if you want to pull the container from neuralet dockerhub)
docker build -f jetson-nano.Dockerfile -t "neuralet/jetson-nano:applications-smart-distancing" .

# 3) Run Docker container:
docker run -it --runtime nvidia --privileged -p HOST_PORT:8000 -v /PATH_TO_CLONED_REPO_ROOT/:/repo neuralet/jetson-nano:applications-smart-distancing
```
for further information please follow the instruction in:
https://github.com/neuralet/neuralet/blob/master/applications/smart-distancing/README.md
let me now if it was helpful :)

grdsvt, 2020-05-20:
Hi, 
thanks alpha-carinae29 for your reply.
I downloaded docker repository with "docker pull neuralet/jetson-nano:applications-smart-distancing" 
and then I tried to run docker with " docker run -it --runtime nvidia --privileged -p HOST_PORT:8000 -v /PATH_TO_CLONED_REPO_ROOT/:/repo neuralet/jetson-nano:applications-smart-distancing
but I received an error message :
"python3: can't open file 'neuralet-distancing.py': [Errno 2] No such file or directory"
I selected the folder path where I see neuralet-distancing.py (if it is the main python file to execute) but I received this message.
I have no experience with docker, sorry.
Can sameone help me?

Thanks

mhejrati, 2020-05-20:
@grdsvt 
you need to replace "/PATH_TO_CLONED_REPO_ROOT/" with full absolute address of your locally cloned repo. e.g. /users/grdvst/projects/neuralet/ 
Right now the container can't see the files because the mounted volume is not valid. 
In the future releases, we are planning to move all the code inside the container to make this simpler and not require you to mount code from outside.

grdsvt, 2020-05-20:
Thanks.
So the code in this repository is not complete?
When do you think to insert all code inside the container?
Why haven't insert all code in github without docker?
Thanks for your reply.

alpha-carinae29, 2020-05-20:
Also please replace "HOST_PORT" with a open port in your local machine. e.g. 8000 and after you ran the docker, open your browser and browse `localhost:8000`

grdsvt, 2020-05-20:
Thanks alpha-carinae29.
I replaced "HOST_PORT" with 8000 but I did not open browser and not browe localhost:8000.
But I don't understand......
The repository for jetson is not complete yey, right?

Thanks

alpha-carinae29, 2020-05-20:
> So the code in this repository is not complete?

The code is complete in the repository, However the codes are not contain in docker container and as @mhejrati mentioned, for now you should mount the source code to the container.

alpha-carinae29, 2020-05-20:
> The repository for jetson is not complete yet, right?

No, it is complete for Jetson and we have a step by step guide for Jetson Nano device [here](https://github.com/neuralet/neuralet/tree/master/applications/smart-distancing#getting-started).
please follow this instruction and let us know if you had any problem.
thanks

grdsvt, 2020-05-21:
Hi @alpha-carinae29.
I executed :

cd neuralet/applications/smart-distancing/

# 1) Download TensorRT engine file built with JetPack 4.3:
./download_jetson_trt.sh
(OK)

# 2) Build Docker image 
docker build -f jetson-nano.Dockerfile -t "neuralet/jetson-nano:applications-smart-distancing" 

"docker build" requires exactly 1 argument.
See 'docker build --help'.
Usage:  docker build [OPTIONS] PATH | URL | -

# 3) Run Docker container:
docker run -it --runtime nvidia --privileged -p HOST_PORT:8000 -v /PATH_TO_CLONED_REPO_ROOT/:/repo neuralet/jetson-nano:applications-smart-distancing

I am in the neuralet/applications/smart-distancing/ path but I received this message on 2 step.
Sorry but I am new on docker......

alpha-carinae29, 2020-05-22:
It seems you forgot the `.` at the end of docker build command. So please try:
`docker build -f jetson-nano.Dockerfile -t "neuralet/jetson-nano:applications-smart-distancing" .`

grdsvt, 2020-05-24:
I solved. 
The problem was that I had jetpack 4.4 and it was impossible importing all dipendencies.
I downgrade to jetpack 4.3 and now it is working.

closed 2020-05-24 by grdsvt