galliot-us/neuralet #111: Generate build artifacts for commits

mhsekhavat, 2020-05-19:
We are currently building two artifacts using docker in `gstreamer` branch:
1) gstreamer (CMake): https://github.com/neuralet/neuralet/blob/2a513f1405c972b82ab5e83d5b02dc794fb98eb2/applications/smart-distancing/Dockerfile-x86-openvino#L33-L43
2) frontend (NPM): https://github.com/neuralet/neuralet/blob/2a513f1405c972b82ab5e83d5b02dc794fb98eb2/applications/smart-distancing/Dockerfile-x86-openvino#L1-L6

These two build processes take a lot of time. We should use CI/CD for building them instead.

mhejrati, 2020-05-19:
@MrUPGrade your comments are appreciated :)

MrUPGrade, 2020-05-21:
Agree, will try to do it once I finish setting up the CI server.

closed 2022-12-12 by mhejrati