galliot-us/neuralet #115: Extract smart-distancing to https://github.com/neuralet/smart-social-distancing

mhsekhavat, 2020-05-20:
Steps:

- [x] move commits
- [ ] fix paths in configs
- [ ] move issues

mhsekhavat, 2020-05-20:
PR for fixing paths: https://github.com/neuralet/smart-social-distancing/pull/2

mhsekhavat, 2020-05-20:
Moved commits using the following scripts:
```
git subtree split -P applications/smart-distancing -b develop
cd ../smart-social-distancing
git init
git pull ../neuralet develop
```

mhejrati, 2020-05-20:
Thanks @mhsekhavat we needed this so badly :D

mhsekhavat, 2020-05-21:
In order to move a branch from this repo to the new repo, you can follow this procedure:

```
OLD_REPO="$PWD/neuralet"  # CHANGE IT to the absolute path of cloned neuralet repo
NEW_REPO="$PWD/smart-social-distancing"  # CHANGE IT to the absolute path of cloned new repo (if it is not already cloned, please clone it first.)
BRANCH='ui-skelton'  # CHANGE IT to the name of the branch you want to move
cd $OLD_REPO
git checkout $BRANCH
git subtree split -P applications/smart-distancing -b "splitted-$BRANCH"
cd $NEW_REPO
git checkout master -b $BRANCH
git pull $OLD_REPO "splitted-$BRANCH"
```

closed 2022-12-12 by mhejrati