galliot-us/neuralet #116: Adding Pedestrian Detection Models for x86

alpha-carinae29, 2020-05-20:
Hi everyone,
I found out that only one model is available for x86 devices and this model is a COCO pre-trained model.
So I decided to train different pedestrian detection model with TensorFlow object detection API. I have trained three different object detector with Oxford Town Center dataset so far and I wanted to upload these models somewhere so I can download them in smart distancing app, however usually these models have large size and GitHub repositories may be not the best platform for uploading them.
@mhejrati what is your suggestions?

mhejrati, 2020-05-20:
We use this repo for that purpose:
https://github.com/neuralet/neuralet-models

Make a PR there and I can merge.

alpha-carinae29, 2020-05-20:
Yeah but some of the model's checkpoints are above 100MB and GitHub have restrictions on large files.

mrn-mln, 2020-05-23:
@alpha-carinae29 instead of checkpoints just add a frozen graph (.pb file) to the repo.

alpha-carinae29, 2020-05-25:
Thanks @mrn-mln, however `.pb` files for faster RCNN models are above 100MB

alpha-carinae29, 2020-05-25:
@mhejrati I pushed two models in here:
https://github.com/neuralet/neuralet-models/pull/6

closed 2020-06-03 by mhejrati