galliot-us/neuralet #118: Smart distancing on Jetson nano has very low fps

grdsvt, 2020-05-24:
Hi, 
I solved the problem on my Jetson nano downgrade jetpack from 4.4 to 4.3 version and it is working,
but I see thet in the video it should have 17 - 20 fps (I read this in the blue text) but seeing the video I suppose that  fps is about 2-3, very slow.
How is it possible? Where is the problem?

Thanks

mrn-mln, 2020-05-24:
Hi @grdsvt,
Our video streaming method is a basic HTML method and this is why you see the output frame rate slow. The actual processing time is what the blue text indicates.
This issue will be solved at #87

wenzuolin-maker, 2020-05-29:
Hi, I have a problem.
I want to open camera on nano using gstreamer. So I changed to "VideoPath: v4l2src device=/dev/video0 ! video/x-raw, width=640, height=480 ! 'videoconvert ! appsink" in config-jetson.ini file. This gstreamer pipline is working in python file.
And I changed "input_cap = cv.VideoCapture(video_uri, cv.CAP_GSTREAMER)" in libs/core.py.
but I saw "failed to load video ".
How can I solve it?

mhejrati, 2020-06-03:
@mhsekhavat

codebugged, 2020-06-29:
Hi @wenzuolin-maker 
Was your problem solved ?

If yes then please help me

JsonSadler, 2020-07-07:
> Hi, I have a problem.
> I want to open camera on nano using gstreamer. So I changed to "VideoPath: v4l2src device=/dev/video0 ! video/x-raw, width=640, height=480 ! 'videoconvert ! appsink" in [...]

more like it has been failed to load the default video sample as it has been removed, can you confirm the problem is with the VideoCapture for sure? 

@codebugged What was your problem exactly?

Some work has been done around to support gstreamer input, but not well tested and merged yet. See #70

closed 2022-12-12 by mhejrati