galliot-us/neuralet #121: ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace:

Lalithmohan24, 2020-05-27:
Hello,

I am using A jetson nano with jetpack 4.3 and trying to run a project from jetson community  " Smart Social Distancing ".

While trying to execute the program i get this error..

[TensorRT] ERROR: Could not register plugin creator:  FlattenConcat_TRT in namespace:
failed to load video /home/dlinano/neuralet/applications/smart-distancing/data/TownCentreXVID.avi

While googling about this error, i got some sites saying, the locally installed tensorflow version is not compatible to work with UFF models.

My questions are:

1) How to solve the error: Could not register plugin creator:  FlattenConcat_TRT in namespace?
2) Why the program is not taking any input video ?

mrn-mln, 2020-05-27:
Hi @Lalithmohan24
The error number 1 is not important the application will run without any issue.
For issue number 2 you must download the sample video file as follows:
```
cd neuralet/applications/smart-distancing/
# Download a sample video file from https://megapixels.cc/oxford_town_centre/
./download_sample_video.sh```
 (or if you want to load your custom video file you should indicate your source video file at config-jetson.ini)
Please read the smart-distancing README carefully.

mhejrati, 2020-06-03:
For FlattenConcat_TRT it is ok to ignore it, it is being fixed in future releases.

closed 2020-06-03 by mhejrati