galliot-us/neuralet #122: Video for testing pedestrian detection generalization

mrn-mln, 2020-05-27:
Hi,
I've found a video which is helpful to test the generalization of trained models on the social-distancing app.
Please keep related topics here and share sample videos here.

https://github.com/opencv/opencv/blob/master/samples/data/vtest.avi

cachro2, 2020-06-29:
Does anyone have a link to the original video: TownCentreXVID.avi

The link is broken:
neuralet/applications/smart-distancing/download_sample_video.sh

wget http://www.robots.ox.ac.uk/ActiveVision/Research/Projects/2009bbenfold_headpose/Datasets/TownCentreXVID.avi -O data/TownCentreXVID.avi

JsonSadler, 2020-07-07:
@mrn-mln  I think you can close this now

@cachro2 Check [https://github.com/neuralet/smart-social-distancing/pull/26](https://github.com/neuralet/smart-social-distancing/pull/26), videos are changed as the original video has been removed due to some privacy issues.

JsonSadler, 2020-07-26:
I think this can be closed now as https://github.com/neuralet/smart-social-distancing/pull/26 is merged and closed @mrn-mln @mhejrati

closed 2022-12-12 by mhejrati