galliot-us/neuralet #123: centroid and fourpoint coordinate

Lalithmohan24, 2020-05-31:
Hello,

I am executing your smart social distancing application i have some doubts in this application.

Questions

1.Difference between centerpointdistance and fourpointdistance explain with pictures?

2.In this application your using NMS concept to avoid duplicate bounding boxes but when i try to give an a video as input there is a duplicate bounding box occuring what is the problem and how to solve it?

[image]

3.How do you calculate the bounding boxes coordinate, centroid coordinate and fourpoint coordinate please give me an image representation for understanding the concept?

4.How to calculate the length of the two bounding boxes based on axis or direction?

emma-w-dev, 2020-06-01:
Hi @Lalithmohan24, 

as it came in parameter description,  `CenterPointsDistance` compare center of pedestrian boxes together and `FourCornerPointsDistance` compare four corresponding corner points of pedestrian boxes and suppose the minimum value of them as people distances. 

Four corner points method supposed to be more sensitive and accurate in smaller scenes. You need to apply both with different `DistThreshold` to find a better combination for your own site. We can help to you to fine-tune parameters on your site if we can access to an instance of your stream. 

There is a detailed tutorial on how the application works in [Neuralet website.](https://neuralet.com/docs/tutorials/smart-social-distancing/)

mhejrati, 2020-06-03:
Hi @Lalithmohan24 

Which device and model are you running this on? 

Some answers below:
2- For the NMS, you can play with the parameters in the config to see if any of them work better or not, however, ultimately the person detection model can be adjusted and fine-tuned to reduce those instances.
3- Let me know if this is not clear enough, happy to clarify further: https://neuralet.com/docs/tutorials/smart-social-distancing/
4- Can you please elaborate?

closed 2022-12-12 by mhejrati