galliot-us/neuralet #124: Pedestrian Detection Datasets

alpha-carinae29, 2020-05-31:
Hi,
We are gathering various surveillance datasets for pedestrian detection for training models and evaluating generalization of different approaches.
please list all of these datasets under this issue, thanks

alpha-carinae29, 2020-05-31:
1. [Oxford Town Center Dataset](https://www.robots.ox.ac.uk/ActiveVision/Research/Projects/2009bbenfold_headpose/project.html)
2. [Caltech Pedestrian Detection](http://www.vision.caltech.edu/Image_Datasets/CaltechPedestrians/)
3. [Virat Video Dataset](https://viratdata.org/)

mhejrati, 2020-06-03:
Is this similar to #122 or different?

alpha-carinae29, 2020-06-03:
Yeah , sorry for duplication!

Darlene1213, 2020-10-30:
can anyone provide me town center dataset please?? As it is unavailable on the link

dreamyt, 2022-11-07:
> can anyone provide me town center dataset please?? As it is unavailable on the link

have you gathered the town center dataset? Because now I also need this

closed 2022-12-12 by mhejrati

Xu-Justin, 2022-12-29:
> > can anyone provide me town center dataset please?? As it is unavailable on the link
> 
> have you gathered the town center dataset? Because now I also need this

Have you found it @dreamyt? because I also need this, hahaha.