galliot-us/neuralet #126: collaborate thru python only

orapradeep, 2020-06-04:
I am not sure about dockers, can i colloborate through python code developement??

mrn-mln, 2020-06-04:
@orapradeep Docker is a convenient service for installing all packages and delivering software, however, if you are not sure about using it, you can install all packages manually and run the application by python.

orapradeep, 2020-06-04:
Thx @mrn-mln any pointers on the structure will be helpful

mrn-mln, 2020-06-04:
@orapradeep 
After Cloning the repo go to ` neuralet/applications/smart-distancing/`
For x86 devices open `config-x86.ini` file and set your config.
Run the app by this script `python neuralet-distancing.py --config config-x86.ini`
Note: For running the app you should install needed packages/frameworks/libraries such as flask,/ tensorflow, numpy ...
Please let me know if you have any issue.

JsonSadler, 2020-07-07:
@orapradeep Thanks for your willingness, feel free to collaborate and make PR's, just mention that you need help with the dockers.