galliot-us/neuralet #127: MIPI camera input

delgadocc, 2020-06-07:
Hello, thank you so much for this amazing job! I'm trying to run the algorithm using my MIPI camera on the Jetson Nano. Can you help me, please? I'm trying to use the following code but it's not working. Somebody already do it?

Thank you

cv2.VideoCapture("nvarguscamerasrc ! video/x-raw(memory:NVMM), " \
    "width=(int)1280, height=(int)720,format=(string)NV12, " \
    "framerate=(fraction)120/1 ! nvvidconv ! video/x-raw, " \
    "format=(string)BGRx ! videoconvert ! video/x-raw, " \
    "format=(string)BGR ! appsink", cv2.CAP_GSTREAMER)

mrn-mln, 2020-06-09:
@delgadocc I think [this](https://www.e-consystems.com/Articles/Camera/opencv-jetson-using-13MP-MIPI-camera.asp) would be helpful.

JsonSadler, 2020-07-07:
@delgadocc Could you manage to solve it?

delgadocc, 2020-07-24:
Not jet!

JsonSadler, 2020-07-26:
Does this help? 
https://github.com/neuralet/smart-social-distancing/pull/34#issuecomment-664015294

closed 2022-12-12 by mhejrati