galliot-us/neuralet #128: Model download not working from git LFS

JsonSadler, 2020-06-19:
**Describe the bug**
Downloading models from [neuralet/neuralet-models](https://github.com/neuralet/neuralet-models) does not work any longer as it seems to use Git LFS.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone, be in master
2. Build docker for amd64 node with usb coral
3. Run as instructed in the readme.
4. You get this error:

`model does not exist under:  libs/detectors/edgetpu/data/ped_ssdlite_mobilenet_v2_quantized_edgetpu.tflite downloading from  https://raw.githubusercontent.com/neuralet/neuralet-models/master/edge-tpu/pedestrian_ssdlite_mobilenet_v2/ped_ssdlite_mobilenet_v2_quantized_edgetpu.tflite
100% [..................................................................................] 132 / 132Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 13, in __init__
    self.engine = CvEngine(self.config)
  File "/repo/applications/smart-distancing/libs/core.py", line 26, in __init__
    self.detector = Detector(self.config)
  File "/repo/applications/smart-distancing/libs/detectors/edgetpu/detector.py", line 24, in __init__
    self.net = pedestrian_ssdlite_mobilenet_v2.Detector(self.config)
  File "/repo/applications/smart-distancing/libs/detectors/edgetpu/pedestrian_ssdlite_mobilenet_v2.py", line 44, in __init__
    self.interpreter = Interpreter(self.model_path, experimental_delegates=[load_delegate("libedgetpu.so.1")])
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 204, in __init__
    model_path, self._custom_op_registerers))
ValueError: Model provided has model identifier 'ion ', should be 'TFL3'
`

**Expected behavior**
I checked the file which was an ASCII, got from the url `https://raw.githubusercontent.com/neuralet/neuralet-models/master/edge-tpu/pedestrian_ssdlite_mobilenet_v2/ped_ssdlite_mobilenet_v2_quantized_edgetpu.tflite`, which is now only a git lfs reference and it clearly does not work. The code has to download the model itself somehow. 

I think this issue is related to all neuralet codes which download any models from neuralet-models repository.

JsonSadler, 2020-06-19:
Seems when you click on the download button in the browser, the link to file is something like: 
https://media.githubusercontent.com/media/neuralet/neuralet-models/master/edge-tpu/pedestrian_ssdlite_mobilenet_v2/ped_ssdlite_mobilenet_v2_quantized_edgetpu.tflite

JsonSadler, 2020-06-23:
Thanks @emma-w-dev

mhejrati, 2020-06-23:
It is fixed for social distancing here:
https://github.com/neuralet/smart-social-distancing/pull/22

We need to update the links for the rest of models.

JsonSadler, 2020-06-23:
But let's keep this issue open as it affects the neuralet models too, not only the smart distancing app.

closed 2022-12-12 by mhejrati