galliot-us/neuralet #129: Jetson Nano Jetpack 4.4

svetlanadataper, 2020-06-29:
**Describe the bug**
I was trying to test the detection algorithm on my Jetson Nano, I followed the instructions for building the container.

**To Reproduce**
Steps to reproduce the behavior:
1. I built the container as stated in the repo:

`sudo docker build -f jetson-nano.Dockerfile -t "neuralet/jetson-nano:applications-smart-distancing" .`

2. When I run I get the following error:

`sudo docker run -it --runtime nvidia --privileged -p 8000:8000 -v /srv/demo/neuralet:/repo neuralet/jetson-nano:applications-smart-distancing`

`Traceback (most recent call last):
  File "neuralet-distancing.py", line 22, in <module>
    DistanceApp(args)
  File "neuralet-distancing.py", line 13, in __init__
    self.engine = CvEngine(self.config)
  File "/repo/applications/smart-distancing/libs/core.py", line 23, in __init__
    self.detector = Detector(self.config)
  File "/repo/applications/smart-distancing/libs/detectors/jetson/detector.py", line 17, in __init__
    from . import mobilenet_ssd_v2
  File "/repo/applications/smart-distancing/libs/detectors/jetson/mobilenet_ssd_v2.py", line 3, in <module>
    import tensorrt as trt
  File "/usr/local/lib/python3.6/dist-packages/tensorrt/__init__.py", line 1, in <module>
    from .tensorrt import *
ImportError: libnvinfer.so.6: cannot open shared object file: No such file or directory`

I assume this is inside the docker container the missing library because I do not have this problem on the Jetson nano itself.

Any help would be appreciated.

Thank you

Svetlana

cachro2, 2020-06-29:
I am experiencing the same issue.

svetlanadataper, 2020-06-29:
> I am experiencing the same issue.

It seems like the container is built with missing libraries...anyone can help us with this?

engineer1982, 2020-07-01:
Same problem here guys... :(

JsonSadler, 2020-07-07:
@svetlanadataper We only support jetpack 4.3 for now, but what you told was a little bit weird, what did you meant by "because I do not have this problem on the Jetson nano itself"? Can you run the code outside the docker? 
I am asking as some of the libraries used in converting our models to use with TRT are build in a way that are not working on Jetpack 4.3 (We had troubles to get it working with 4.4: [https://github.com/neuralet/neuralet/issues/72](https://github.com/neuralet/neuralet/issues/72) )

JsonSadler, 2020-07-07:
@cachro2 @engineer1982 Are you using Jetpack 4.4 too? 
There should be no issues with 4.3

engineer1982, 2020-07-07:
I downgraded my Jetson Nano to JetPack 4.3 and worked fine. Thanks!

svetlanadataper, 2020-07-07:
> @svetlanadataper We only support jetpack 4.3 for now, but what you told was a little bit weird, what did you meant by "because I do not have this problem on the Jetson nano itself"? Can you run the [...]

I understand, probably the problem is Jetpack 4.4. I cannot downgrade now so hopefully in the future it will work on 4.4.

Thank you

Svetlana

JsonSadler, 2020-07-07:
> I understand, probably the problem is Jetpack 4.4. I cannot downgrade now so hopefully in the future it will work on 4.4.
>
> Thank you
>
> Svetlana

There are some work in progress under deepstream-gstreamer branch, @mdegans may have some ideas that if it helps running on 4.4 using deepstream or not. I was away for a long time and I am not aware of the exact situation, but you can see [previous discussions around this ](https://github.com/neuralet/neuralet/issues/72#issuecomment-627470444). You also can join our slack and get the updates from collaborators. 

I am not closing this issue for now, until the support is ready.

mdegans, 2020-07-07:
This branch runs on 4.4, @svetlanadataper 
https://github.com/mdegans/smart-social-distancing/tree/gst-merge

Just waiting for a code review so it can be merged into master. No guarantees it will be accepted since the way it works differs from the other branches, but it is something you can use right now. Check out the readme in that repo for instructions on building/running.

svetlanadataper, 2020-07-07:
> This branch runs on 4.4, @svetlanadataper
> https://github.com/mdegans/smart-social-distancing/tree/gst-merge
>
> Just waiting for a code review so it can be merged into master. No guarantees it will be [...]

Thank you so much! I will test it! I appreciate it...

Svetlana

ycRose, 2020-07-14:
> This branch runs on 4.4, @svetlanadataper
> https://github.com/mdegans/smart-social-distancing/tree/gst-merge
>
> Just waiting for a code review so it can be merged into master. No guarantees it will be [...]

hi, I'm trying jetson xavier nx, it has jetpack4.4 and tensorrt7.1.0, can I use this branch? Actually I tried yesterday, but failed with some xxx.so error like 'ImportError: libcurand.so.10.0'. Do I need to modify this branch and run? looking forward your reply

mdegans, 2020-07-15:
> hi, I'm trying jetson xavier nx, it has jetpack4.4 and tensorrt7.1.0, can I use this branch? Actually I tried yesterday, but failed with some xxx.so error like 'ImportError: libcurand.so.10.0'. Do I need to modify this branch and run? looking forward your reply

If you're using 4.4 GA, that's untested and there's been some breakage. Officially DeepStream is not supported on it until the GA release of DeepStream itself. I am working on getting it building and running as it did under 4.4 DP.

No ETA as of yet, as I am working on a separate social distancing project RN. Hopefully I'll be able to integrate the changes from that into the gst-merge branch of smart-social-distancing by the end of the week but there are no promises.

If you're running under the developer preview, it should work if you do `deepstream.sh build`, edit your deepstream.ini as indicated in the README.me and then `deepstream.sh run` if it doesn't, please report any error you get here.

`libcurand.so.10.0` sounds a bit odd since that's the version included with JetPack 4.3. Jetpack 4.4 and DP have cuda 10.2.

ycRose, 2020-07-16:
> > hi, I'm trying jetson xavier nx, it has jetpack4.4 and tensorrt7.1.0, can I use this branch? Actually I tried yesterday, but failed with some xxx.so error like 'ImportError: libcurand.so.10.0'. Do I need to modify this branch and run? looking forward your reply
> 
> @ycRose
> 
> If you're using 4.4 GA, that's untested and there's been some breakage. Officially DeepStream is not supported on it until the GA release of DeepStream itself. I am working on getting it building and running as it did under 4.4 DP.
> 
> No ETA as of yet, as I am working on a separate social distancing project RN. Hopefully I'll be able to integrate the changes from that into the gst-merge branch of smart-social-distancing by the end of the week but there are no promises.
> 
> If you're running under the developer preview, it should work if you do `deepstream.sh build`, edit your deepstream.ini as indicated in the README.me and then `deepstream.sh run` if it doesn't, please report any error you get here.
> 
> `libcurand.so.10.0` sounds a bit odd since that's the version included with JetPack 4.3. Jetpack 4.4 and DP have cuda 10.2.
Hi, thanks for your reply, my deepstream is 4.4 DP and I tried it, some problems:
1. **.git is not found**
    I fixed it by 'git init';

2. **fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'**
    I can't fix it, so I assign 'GIT_BRANCH=deepstream' directly,and during building: warnings in red like:
'npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@1.2.13', passed it and I get 
Successfully built dbdc196e1293
Successfully tagged neuralet/smart-distancing:deepstream-0.1.0-aarch64
Successfully tagged neuralet/smart-distancing:deepstream-deepstream-aarch64;

3.during run:
**ERROR**:Primary_Detector/resnet10.caffemodel_b30_gpu0_int8.engine open error
**WARN**:deserialize backend context from engine from file /Primary_Detector/resnet10.caffemodel_b30_gpu0_int8.engine failed, try rebuild
**WARN**:failed to serialize cude engine to file: /Primary_Detector/resnet10.caffemodel_b2_gpu0_int8.engine
...
**ERROR**: Serialize engine failed because of file path: /opt/nvidia/deepstream/deepstream-5.0/samples/models/Primary_Detector/resnet10.caffemodel_b2_gpu0_int8.engine opened error
**WARN**:failed to serialize cude engine to file: /opt/nvidia/deepstream/deepstream-5.0/samples/models/Primary_Detector/resnet10.caffemodel_b2_gpu0_int8.engine
...
**ERROR**:GstEngine:gst-resource-error-quark: Error while writing to file. (10): gstmultifilesink.c(810): gst_multi_file_sink_write_buffer (): /GstPipeline:pipeline0/GstBin:bin0/GstHlsSink:hlssink0/GstMultiFileSink:multifilesink0:
No such file or directory
...
GstEngine:shifting pipeline to NULL state.

    I checked Primary_Detector folder and there is no engine file,so I build incorrectly? or rebuild engine failed?
can I copy my engine file and cinfig file into image to solve? looking forward your reply:)

mdegans, 2020-07-26:
```
ERROR:GstEngine:gst-resource-error-quark: Error while writing to file. (10): gstmultifilesink.c(810): gst_multi_file_sink_write_buffer (): /GstPipeline:pipeline0/GstBin:bin0/GstHlsSink:hlssink0/GstMultiFileSink:multifilesink0:
No such file or directory
```
@ycRose , @mats-claassen 

This likely means the path specified for the hlssink to write to does not exist. I believe that may be because it's an empty path and git does not allow empty paths. I don't maintain this branch, but it should be as simple as creating the folder and putting a .keep file (or similar trick) to ensure it does exist when cloned. Such a change should probably be made in the other (new) repo, and this one locked.

closed 2022-12-12 by mhejrati