galliot-us/neuralet #131: Have trouble understanding the instructions to implement this project

tanueihorng, 2020-07-13:
Hi, I'm sorry if you find this thread funny. I'm a secondary sch student with no experience or prior knowledge on this. 
I was given a jetson nano for object detection project but I wanted to add smtg valuable to the project for our school.
As COVID-19 virus cases keep increasing in the community, I would like to try it out in our school canteen. 

I had followed the instructions in the homepage, but couldn't get it working.  
I'm currently using Jetson Nano and already successfully implemented the object detection program by cloning Jetson-Inference by dusty from Nvidia (SSD_Mobilenet_V2). Can i use RTSP stream (IP Camera) as an input ?

Please help me out! 
[image]
[image]

mats-claassen, 2020-07-13:
Hi @tanueihorng, you should be able to run it on IP cameras by changing the `VideoPath` in the `config-jetson.ini`. You should point it at the URL of the camera.

I also see that you did not replace the placeholders in the docker run command: `HOST_PORT` and `PATH_TO_CLONED_REPO_ROOT` should be replaced by the correct values. For HOST_PORT choose any free port where you want to access the Dashboard, like 8000.

tanueihorng, 2020-07-13:
Thank you for your fast reply! 

docker run -it --runtime nvidia --privileged -p 8000:8000 -v /nueralnet/applications/smart-distancing/:/repo neuralet/jetson-nano:applications-smart-distancing

Is it smtg like this? I replaced the PATH_TO_CLONED_REPO_ROOT to the cloned file's path

tanueihorng, 2020-07-13:
After changing the VideoPath in the config-jetson.ini, just save it ? or do I need to do something else?
So the program is actually triggered by launching the above docker run command line right?

Really appreaciate your help!

mats-claassen, 2020-07-13:
Yes, the docker run command starts the program. If you modified the config-jetson.ini file then you have to rebuild the docker image with `docker build` (have a look at the readme)

Personally, I haven't tried it with an IP camera but you should only have to save the config file and that is it. Unless the camera requires some username/password

tanueihorng, 2020-07-14:
Hi @mats-claassen 

I have tried the docker run command but it returned this error. Any idea why this error occurred?
[image]

Also If I wan do it with IP Camera, just put in the RTSP URL then save it right ? (Yes, my camera required username/password)
[image]

mats-claassen, 2020-07-14:
First, check the path to the repo you used in the first command. It has some typos. Might be the issue for your error.

And for VideoPath, yes, just set the URL and save it. However in the image you posted you have added a whole `rtspsrc` command. You should add just the URL. There are several posts on StackOverflow which could help you get the correct URL like [this one](https://stackoverflow.com/a/42773860/4708657). As said I haven't tried it with an IP camera yet.

And be sure to be on the latest master commit for best feature set

tanueihorng, 2020-07-16:
Hi @mats-claassen 
I have corrected the RTSP URL and some typos but it seems not working also
[image]

mats-claassen, 2020-07-16:
The Readme says you only need this volume: `"$PWD/data":/repo/data`. You must run that from your `smart-distancing` folder and it should work. `PWD` is an alias to the absolute path of your current directory

tanueihorng, 2020-07-17:
Tried it but this time new error occurred 
[image]

mats-claassen, 2020-07-17:
As the error says: There seems to be no `Host` in the `App` section of the config file you are using. Or maybe the config-jetson.ini file is not where it should be. You are still not setting the volume as I mentioned above (also as explained in the Readme). Maybe that is the issue.

tanueihorng, 2020-07-17:
Hi @mats-claassen 
I'm really sorry! Do you mind explaining what is "$PWD/data":/repo/data

mats-claassen, 2020-07-17:
It seems you have cloned this repo. The smart social distancing app has been moved to https://github.com/neuralet/smart-social-distancing. So please start by cloning that.

If you follow the [README here](https://github.com/neuralet/smart-social-distancing#usage) it says you should run the following commands for Jetson Nano:

```bash
# 1) Download TensorRT engine file built with JetPack 4.3:
./download_jetson_nano_trt.sh

# 2) Build Docker image for Jetson Nano (This step is optional, you can skip it if you want to pull the container from neuralet dockerhub)
docker build -f jetson-nano.Dockerfile -t "neuralet/smart-social-distancing:latest-jetson-nano" .

# 3) Run Docker container:
docker run -it --runtime nvidia --privileged -p HOST_PORT:8000 -v "$PWD/data":/repo/data neuralet/smart-social-distancing:latest-jetson-nano
```

As you see, the volume in the `docker run` command is `"$PWD/data":/repo/data`. That is what you should do as well. And remember that `HOST_PORT` is a placeholder so replace it with the port you want to use.

tanueihorng, 2020-07-17:
I think my current version of Jetpack is 4.4. Does it work on 4.4?

mats-claassen, 2020-07-17:
Not yet. See #129 for updates on that

JsonSadler, 2020-07-26:
@tanueihorng Could you solve your problem? Be aware that the current master branch of the new repo is experiencing heavy changes (Working, of course). The readme is not yet updated (but you can find it here: https://github.com/neuralet/smart-social-distancing/blob/c00822b503aff84bb3fae8663c95c89a04ba7767/README.md) 

Let me know if you still have troubles running the code, assuming you have installed Jetpack 4.3 now.