galliot-us/neuralet #141: Module not found in docker image for object detection training

FSet89, 2021-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Build and launch docker container from source according to [this](https://neuralet.com/article/quantization-of-tensorflow-object-detection-api-models/) guide
2. run python object_detection_model_main.py --help

**Expected behavior**
The python script should run and display help page

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04

**Additional context**
When trying to run the script, I got errors for missing modules:

```
root@89b340f3a5bf:/models/research# python object_detection/model_main.py 
Traceback (most recent call last):
  File "object_detection/model_main.py", line 25, in <module>
    from object_detection import model_lib
  File "/models/research/object_detection/model_lib.py", line 27, in <module>
    import tf_slim as slim
ModuleNotFoundError: No module named 'tf_slim'
```
Same with lvis.

alpha-carinae29, 2021-03-19:
Hi @FSet89, 
Can you please test one more time with this modification?

FSet89, 2021-03-19:
Hi, actually I fixed by pulling the container form docker hub (2nd option), thank you anyway

mhejrati, 2022-12-12:
PR #141 addresses this

PR #142 (2022-12-12): update tf object detection api docker file
This will close #141
closed 2022-12-12 by mhejrati